MaikStohn/UP3D #1: first test

Gitty-Hub, 2016-03-11:
Hi,thanks for all this work.

But I did not get it to work. On 64 bit Windows 10 I get the error it's not compatible (32 bit).

Then I tried 32 bit Win XP but I getan error trying to ranscode: something with 16 bit DOS invalid instruction...

Can somebody with Windows help or will this be fixed with the new GUI?

MaikStohn, 2016-03-24:
Hi, I tested the compiled windows binary on Windows 7 32 and 64 bit and Windows 10 64 bit without any problems.

Please make sure that you run the program from command prompt "CMD".

More details here: http://stohn.de/3d/index.php/2016/03/10/up3dtools-little-how-to

Please check again and close the issue in case this solves your problem.

Gitty-Hub, 2016-03-24:
Sorry tried again but I get ''unsupported 16bit application'' and that up3dtranscode.exe isn't supported on 64 bit Windows. I downloaded the .exes again to make sure.. Maybe I'll wait a bit, this is being worked on right? Thanks for everything!

MaikStohn, 2016-03-25:
I really would like to work on it but your problem can not be verified.

It was tested from me on several Windows versions XP-32bit,Win7-32Bit,Win7-64Bit,Win8.1-32Bit,Windows 10-64Bit always with the same result: It works perfectly.

So somehow your configurations seems to be different.

1)
Can you please confirm that you tried running the .exe from the command prompt "CMD" (black window where you must type text).

2)
Please give more information of your windows version:
a) what is the output when you type:   VER   and press return (also in CMD window)
b) what is the output when you type:   SET   and press return (please remove personal information before posting)

Gitty-Hub, 2016-03-25:
Yes, I tried in dosbox/cmd...

Windows version 10.0.10586

MaikStohn, 2016-04-01:
I tried hard to find any resource which describes a similar problem you describe but nothing came up.

The only dependency required from the compiled executable is the standard vc runtime from microsoft. You should have this on your computer as soon as you installed any other software to windows 10.

But maybe just try to install an up to date version:

https://www.microsoft.com/en-US/download/details.aspx?id=48145

In case you still have this problem a screen shot of the exact error message would be very helpful.

MaikStohn, 2016-04-06:
One person reported to me that by mistake he downloaded the .exe file wrong. Maybe this is also the problem for you.

==> Do NOT use right click "Save As..." when clicking on the .exe file to download it.

Instead:
- click on the .exe file name ==> a new site comes up telling you that it is binary content
- click on the "RAW" button and the download will start

Hope this was also your problem.

Gitty-Hub, 2016-04-07:
That worked! Would have never guessed that RAW button.. Is that a noob mistake? What's different when you download the first .exe?

MaikStohn, 2016-04-07:
Hi,

when you right click on the link it will save the content of the next web page (HTML with a lot of text and also having the RAW button).

Nozzle height can be given as 3rd parameter to up3dtranscode now.
e.g.
up3dtranscode input.gcode output.umc 122.35

Since I don't have Up Box I'm not sure if the parameter for motors (speed and distances) are good. You can try but I guess some more work is needed.

Gitty-Hub, 2016-04-08:
OK will try and report back.. By the way, I did have the exe files but the way you described worked better :-)

MaikStohn, 2016-04-10:
Good that it was solved.

closed 2016-04-10 by MaikStohn