MaikStohn/UP3D #3: Flipped print

kscheff, 2016-03-22:
After transcode and uploading the printed object is flipped on either x or y axis.

MaikStohn, 2016-03-24:
The problem comes from the coordinate system definition from tiertime. It is still left handed but rotated...
In next version I will change it to match to the standard left handed coordinate system used by slic3r cura and simplify3d.

kscheff, 2016-03-24:
Before swapping the x/y axis the resulting printed object is mirrored (eg can't read printed letters). After my hack the object prints congruent and I believe it was also correctly oriented (rotation wise) on the build platform.

Before I swapped the axises I inverted all X values, by applying 120.0-x to all assignments and positioned the home to x=120. The result was a non-mirrored print, but wrongly rotated on the build platform.

If you say tiertime uses left hand system, then I believe there must be another x/y swap happening in the chain after the g-code interpreter. I observed the 0,0,0 point is front left. 

Could it be that the indexes for X_AXIS and Y_AXIS in up3dconfig.h should be swapped to match the machine?

MaikStohn, 2016-03-25:
Actually the indices of the axis do match exactly the UP machine and software. When you open the original UP software and load a model you can see the coordinate system on the bottom left of screen.

In next release I will just adapt the more commonly used coordinate system like used in any RepRap. 

BTW: Thanks for testing.

kscheff, 2016-03-25:
Comparing to the Up Software, you are right. The object is placed differently, looks like 120-x is applied.

When I am looking at Slic3r and use the x axis (red) as left/right looking to the front of the UP mini, then swapping x/y axis in your program produces the exact same orientation as the preview when printed :) This is in line with my CAD software (FreeCAD). So I am fine with the axis swap. I see that in the source code you are applying -X in some case, so I am puzzled what the real machine orientation is.

I have seen a lot of inverting bits in my professional career (graphics & video chips), flipping around a bits was sometimes helpful  and often required ;) each block designer was right and wrong at the same time...

MaikStohn, 2016-04-01:
Fixed with commit ec0fbd4d71555c5c7c79130405db852be2d0c084

closed 2016-04-01 by MaikStohn