MaikStohn/UP3D #9: Maximum Filesize

MasterOfDesaster666, 2016-04-09:
I got a weird issue, but first things first:

UP!Mini, purchased at Tchibo in February
UP-Software 2.18
ROM V6.110
FW V3.34 
Slicer: Simplify3D

Every time i transcode and upload a file with a size over 30 MB (did'nt test between 20 and 30 MB), the upload results either in a SD Card Failure or it uploads, starts printing and the nozzle runs into the clips at the side of the build platform (see attachments). Nearly got a heart attack hearing the sound it makes ^^ 

Tried it with the Predator Bust HD (failes), TIE Fighter (failes) but e. g. the Buddha works from thingiverse. The file size of the Buddha.gcode/umc is 15.868 KB/15.014 KB. Predator is 47.265 KB/49.104. On the other hand, the TIE file is 6.139 KB/10.177 KB.

While uploading the %-Scale jumps from 49% to -50% to 7%.

c:\UP3DTOOLS>up3dtranscode predator4.gcode predator.umc 121.39
Height: 112.37mm / Layer: 560 / Time: 13h:30m:43s
Nozzle Height: 121.39mm

c:\UP3DTOOLS>upload predator.umc
UPload: 49103.91 kB sent (14%) [114.45 kB/sec]]UPload: 35229.38 kB sent (-13%) [117.80 kB/sec]
ERROR: Write to SD not finished

So i transcoded and uploaded again:
c:\UP3DTOOLS>upload predator4.umc
UPload: 49049.10 kB sent (14%) [118.67 kB/sec]]UPload: 48885.47 kB sent (14%) [118.72 kB/sec]

Other files i transcoded and uploaded are working flawless. Great tool btw :)

So is it an error while uploading, in the stl itself (UP Software prints it normal), transcode or me?

[image]
[image]

MaikStohn, 2016-04-09:
Hi,

this looks like a calculation error in uptranscode (depends on speed and segment length). 

Can you ZIP and attach the gcode file from the picture?
This will help me to debug the problem.

The upload percentage is just a display flaw for large files. I will correct it.
Upload checks if all data was uploaded correctly (checks block count written to SD card) before it starts the print job.

MasterOfDesaster666, 2016-04-09:
Thanks for the answers and the help 

[predator.zip](https://github.com/MaikStohn/UP3D/files/211354/predator.zip)
[tie.zip](https://github.com/MaikStohn/UP3D/files/211355/tie.zip)

:)

Attached the files below

MaikStohn, 2016-04-10:
Thanks for the files.

Like guessed before it is a calculation error.

I used the debug program "convg" from folder "UP3DPARSE":

`convg predator.umc >predator_umc_reversed.gcode`

And then opened the file with a gcode viewer (Simplify3D in my case).

Result shows that UPMini did exactly what it was told to do (no upload error):

[image]

[image]

I will try to fix it and also make a detection for this corner cases so in case there is a mistake in calculation up3dtranscode should at least WARN about it.

MaikStohn, 2016-04-10:
I fixed it and uploaded new version.

closed 2016-04-10 by MaikStohn