MaikStohn/UP3D #11: Layers are not aligned

yanivg101, 2016-04-11:
Hi,
I get this weird behavior on my UP2 Plus 3DP.
I tried to 3D Print the "Marvin_KeyChain" model, both with Cura and Simplify3D, Change parameters etc but with both of them I got this behavior:
[image]
[image]
It's need to be like this:
[image]

on the UP Software, the print go out well as usual.

All file (STL, GCODE, Settings etc.) is here:
[Debug 11-04-2016.zip](https://github.com/MaikStohn/UP3D/files/212771/Debug.11-04-2016.zip)

MaikStohn, 2016-04-11:
Hello,

after inspecting your UMC file I can not find any issues:

[image]

Did it hit previous layer(s) with the nozzle while printing?

E.g. if first layer pushes to much material then it might get higher than the layer is planned. When the next layer is printed the nozzle might hit this previous layer and need some extra force.

One thing could be that "Up Plus 2" has somewhat smaller force motors / stepper driver / whatever.

-> In this case the maximum acceleration and maximum speed can be reduced ("up3dconf.h") to solve this problem.

Another possibility could be that steps/mm is different on "UP Plus 2"

-> Could you physically measure if the printed object dimensions (X/Y) match your model? 
In case it is much bigger or much smaller (steps/mm wrong) it automatically leads to wrong acceleration / speed values. In this case please try to give as much as accurate measurements as you can. 
Like:  X: model: 12.34mm, X: measured: 8.54 mm

yanivg101, 2016-04-12:
I 3D print it again an d this is what I got (compare to printing with UP Software)
[image]
[image]
[https://youtu.be/fin0ywL7Vas](https://youtu.be/fin0ywL7Vas)

MaikStohn, 2016-04-12:
Thanks for the very great investigation.

From the video and your measurement it's getting obvious what's wrong. 
-> Steps/mm is different for UP Plus 2 (lower value)
Because of this the relative positions (used for printing and movements inside of a layer) lead to wrong/different values than absolute positions (which must be used for layer change / Z movements).

In the video you can see that printer starts with first layer to go to a specific position. But before extrusion starts it rapidly moves to another position (this should not happen).
 
Good news, we can fix this easily :-)

I will create a small little extra utility which will read the machines settings from any connected printer.
I already documented big parts of this settings block some time ago in FixUp3D. I'm sure we can get all printer specific parameters from this and add an extra option for machine type in the transcoder.

yanivg101, 2016-04-12:
I can test it on UP BOX 3D Printer, do you need it?
(I work as a technician for all UP 3D Printer...)

MaikStohn, 2016-04-12:
@yanivg101 This is very kind of you.

I created a new small tool called "upinfo" which will read the printer info from UP printer over USB directly.

- connect printer and turn on
- execute "upinfo"

The output should look like this:

```
UP 3D PRINTER INFO:
TypeID:2778 Serial:215150 ROM:6.1100 Model:UP Mini(A) U1:1.00 U3:1 U4:102 U7:100
Max-X:-120.000000 Max-Y:120.000000 Max-Z:130.000000 Steps/mm-X:854.000000 Steps/mm-Y:854.000000 Steps/mm-Z:854.000000 Steps--A: 40.000000 Print sets: 4
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 1"
  ND:0.30 LT:0.20 SW:0.50 ST:1.00 HW:0.52 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:35.00 
  HV:45.00 XV:50.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.00 
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#1  Name:"Set 2"
  ND:0.30 LT:0.25 SW:0.53 ST:1.00 HW:0.55 HS:6.00 HL:3.00 XW:0.57 XS:8.00 XL:3.00 SV:40.00 
  HV:50.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.33 
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#2  Name:"Set 3"
  ND:0.30 LT:0.30 SW:0.55 ST:1.00 HW:0.57 HS:6.00 HL:3.00 XW:0.60 XS:8.00 XL:3.00 SV:40.00 
  HV:50.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.65 
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#3  Name:"Set 4"
  ND:0.30 LT:0.35 SW:0.60 ST:1.00 HW:0.63 HS:6.00 HL:3.00 XW:0.66 XS:8.00 XL:3.00 SV:40.00 
  HV:50.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.10 
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
```

This will help me to find the differences (most likely steps/mm).
Right now I only know for sure the data for many kind of UPMini.

yanivg101, 2016-04-14:
I will upload some info here:

UP 3D PRINTER INFO:
TypeID:277e Serial:406603 ROM:6.1100 Model:UP Plus 2(A) U1:1.00 U3:1 U4:102 U7:100
Max-X:-140.000000 Max-Y:140.000000 Max-Z:140.000000 Steps/mm-X:644.000000 Steps/mm-Y:644.000000 Steps/mm-Z:644.000000 Steps--A: 40.000000 Print sets: 6
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 0"
  ND:0.30 LT:0.15 SW:0.47 ST:1.00 HW:0.50 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:30.00 
  HV:40.00 XV:40.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:0.75 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#1  Name:"Set 1"
  ND:0.30 LT:0.20 SW:0.50 ST:1.00 HW:0.52 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:35.00 
  HV:45.00 XV:50.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.00 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#2  Name:"Set 2"
  ND:0.30 LT:0.25 SW:0.53 ST:1.00 HW:0.55 HS:6.00 HL:3.00 XW:0.57 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.33 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#3  Name:"Set 3"
  ND:0.30 LT:0.30 SW:0.55 ST:1.00 HW:0.57 HS:6.00 HL:3.00 XW:0.60 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.65 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#4  Name:"Set 4"
  ND:0.30 LT:0.35 SW:0.60 ST:1.00 HW:0.63 HS:6.00 HL:3.00 XW:0.66 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.10 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#5  Name:"Set 5"
  ND:0.30 LT:0.40 SW:0.65 ST:1.00 HW:0.68 HS:6.00 HL:3.00 XW:0.72 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.60 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00

UP 3D PRINTER INFO:
TypeID:2779 Serial:505610 ROM:6.1100 Model:UP BOX(M_A) U1:1.00 U3:1 U4:101 U7:100
Max-X:6.000000 Max-Y:-3.000000 Max-Z:210.000000 Steps/mm-X:644.000000 Steps/mm-Y:644.000000 Steps/mm-Z:3220.000000 Steps--A: 40.000000 Print sets: 7
U2:213.00 U3:-262.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 01"
  ND:0.30 LT:0.10 SW:0.45 ST:1.00 HW:0.45 HS:4.00 HL:3.00 XW:0.47 XS:6.00 XL:3.00 SV:30.00 
  HV:40.00 XV:40.00 JS:150.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.45 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#1  Name:"Set 0"
  ND:0.30 LT:0.15 SW:0.47 ST:1.00 HW:0.50 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:30.00 
  HV:40.00 XV:40.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:0.75 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#2  Name:"Set 1"
  ND:0.30 LT:0.20 SW:0.50 ST:1.00 HW:0.52 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:35.00 
  HV:45.00 XV:50.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.00 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#3  Name:"Set 2"
  ND:0.30 LT:0.25 SW:0.53 ST:1.00 HW:0.55 HS:6.00 HL:3.00 XW:0.57 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.33 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#4  Name:"Set 3"
  ND:0.30 LT:0.30 SW:0.55 ST:1.00 HW:0.57 HS:6.00 HL:3.00 XW:0.60 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.65 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#5  Name:"Set 4"
  ND:0.30 LT:0.35 SW:0.60 ST:1.00 HW:0.63 HS:6.00 HL:3.00 XW:0.66 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.10 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#6  Name:"Set 5"
  ND:0.30 LT:0.40 SW:0.65 ST:1.00 HW:0.68 HS:6.00 HL:3.00 XW:0.72 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.60 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00

MaikStohn, 2016-04-14:
Great,

Exactly like I thought: 
X/Y: 644 steps/mm for UP Plus2. Build size:  140x140x140.
compared to 
X/Y: 854 steps/mm for UP Mini. Build size: 120x120x130.

I will add a machine type option to up3dtranscode and then it should work fine for you

However UP box is a bit more different. 1st the coordinate system seems to be translated (X/Y swapped).
And the encoding for maxy/maxy seems to be different:
X/Y: 644 steps/mm for UP Box. Build size: ???x???x210.
(from datasheet it says 255x205x205, so I guess it's 260x210x210)

Let me change the up3dtranscode first and then please try to make a test print on the up box (something with a text so we can verify orientation).

adrianbear, 2016-04-14:
Hi Maik,
I'm online now too, if you would like some immediate testing on an UPBox
Adrian

MaikStohn, 2016-04-14:
@adrianbear Thanks for the offer. Give me some time to make the required changes.

adrianbear, 2016-04-14:
No worries, To be honest I am so excited about this, I plan to stay up all night if need be :)

adrianbear, 2016-04-14:
Do you think the UPBox will print layers less than 0.1 mm high?

MaikStohn, 2016-04-14:
@adrianbear 
Every UP-Printer can print <0.1. But the positioning also needs some precision. Older UP printer are using a pulley and a belt for Z positioning which theoretically allows resolutions down to 0.05 mm.

Since UP-Box Z steps/mm is so much higher (seen from upinfo) than on older UP printer I suspect it is using a lead screw. With this it should be possible to position even much smaller Z distances.

So yes, for sure it can print <0.1 down to 0.01 ... BUT finding the right settings and get a very good bed leveling and ... will take you a lot of tuning time to make it work right.

Small note: Yesterday and today I spent a lot of time with my friend to add the automatic build and release system for the software (every change in software will have automatic checks now and releases are built automatically). So I think there will be no new version today or tomorrow. Sorry.

MaikStohn, 2016-04-15:
I added machine type selection: "mini / classic / plus / box" to up3dtrancode using the info you provided from your machines (Release v0.2.0)

UP-BOX: I'm not 100% sure if I understood the "up box" coordinate system correctly. So please try to print a small object in the middle of the bed first. The object should not be symmetric so we can see if it is mirrored somehow. 

Releases are built automatically and can be found by clicking on "Releases" (top of project site) or by following the link "Download latest" (in the README text on project site).

Please report back.

adrianbear, 2016-04-16:
Hi Maik,

Unfortunately I can't test now. I have had to drive 8 hours from home to see my Mum after a bad car accident. Might be a couple days.

Sorry

Adrian Bear
P: 0403349647
E: abear@live.com.au

https://github.com/MaikStohn/UP3D/issues/11#issuecomment-210665207

adrianbear, 2016-04-21:
Hi Maik,
I think the issue of the transcoder not working on the UPBox might need to be a separate issue because I don't think the issue is related to the transcoding of gcode to upCode in itself.

I have just tested the latest version and added the "box" parameter to the command call.
I have bundled the following calls into a single batch file (called "print.cmd").

I get the same behaviour regardless of input file (the Up Bunny, my orientation test piece or an empty gcode file)
I have recorded the results in a video here => [https://www.wetransfer.com/downloads/23d3432cbaa1fd5ed267651555918b5320160421033916/a9b881]

In short, I am getting the same results as the first version, the print head runs to the front of the printer and grinds the drive motor. At which point I turn the printer off at the switch.
Is there some preliminary content the upload.exe sends to the printer before it starts on the UpCode content?
I have also uploaded my files for you perusal.

[Print.cmd.txt](https://github.com/MaikStohn/UP3D/files/229167/Print.cmd.txt)
[upinfoOutput.txt](https://github.com/MaikStohn/UP3D/files/229168/upinfoOutput.txt)
[Input.gcode.txt](https://github.com/MaikStohn/UP3D/files/229165/Input.gcode.txt)
[output.umc.txt](https://github.com/MaikStohn/UP3D/files/229164/output.umc.txt)

Thanks
Adrian

adrianbear, 2016-04-21:
Hi Maik,

I corrected a minor formatting issue in the batch file (regarding the space before the Nozzle height).
But I get the exact same results as before.
[image]

MaikStohn, 2016-04-21:
Hello Adrian,

thanks for your video. I can see what's going on: The homing procedure is running in wrong direction for UP box. This should be easy to fix.

Some questions:

* before v0.2.0 the print head was also going to front or it changed between v0.1.0 and v0.2.0?

* could you do another video of the normal homing (when you initialize the box). 
  I want to see which axis going to which direction, so I can compare it.

Thanks,

Maik

adrianbear, 2016-04-22:
Hi Maik,

The UPBox has performed the same action for all versions so far.
I have recorded two videos for you that are available in a zip here => [https://we.tl/hLS6T13HwW]
The first video is the UPBox initialization process, the second video is what the UPBox does when it actually starts to print.

If there is anything else I can do to help, please let me know?

Thanks
Adrian

MaikStohn, 2016-04-22:
Hello Adrian,

Thanks for the videos. Based on this I changed homing procedure and coordinate system for the UP-Box (seems is rotated and mirrored).

I just pushed a new version "v0.3.0" which includes all changes. I really hope it just works now for UP-Box. If not then I need a video (again).

Maik

adrianbear, 2016-04-23:
Hi Maik,

The transcoder is working on the UPBox :)
Print orientation is good, I have been printing 3d numbers (2 and 5) as test pieces and placing them off the centre of the bed.
I still need to tune the print settings on Simplify3d a bit and will post more results, but currently my most successful settings are...

Nozzle height used in transcoder is 233.71 (is 208.12 in UP Software)
In Simplify3d my FFF Settings are...
Nozzle=>Nozzle Diameter = 0.3mm
Nozzle=>Extrusion Multiplier = 0.25
Nozzle=>Extrusion Width = Auto
Layer=>First Layer Height = 100% (although I may decrease this)
Layer=>First Layer Width = 120% (although I may increase this)
Temperature=>Heated Bed = 90 degrees (for ABS)
Temperature=>Extruder 1=>Layer 1 = 260 degrees (for ABS)
Temperature=>Extruder 1=>Layer 2 = 250 degrees (for ABS)
Temperature=>Extruder 1=>Layer 3 = 240 degrees (for ABS)

All other settings are default.
I have used these settings to print ABS at 0.2 and 0.1 mm layer heights quite successfully, although they were very small test prints and were removed from the bed (and raft) perhaps a little too easily.
I haven't checked part size accuracy yet and expect there to be some tuning required to manage the ooze/stretch factor and get exact dimensions printed.

It does seem a bit slow to prepare for printing.
Heating the bed from cold takes 10+ minutes, which is not normal when using the UP Software.
Sometimes, it heats the bed but doesn't do anything else. I haven't quite worked out the repeatable pattern here.
Sometimes it seems to wait for the nozzle to cool to less than 45 degrees, before heating it up to temperature again when preparing to print.
At the end of a print it just stops and does not lower the bed, but I'll fix this with some custom gcode script.
The Initialization procedure works fine, but is a bit more brutal than the UP Software. It has quite an audible clunk as it hits the end stops on the X and Y axis (Z seems fine).The UP Software Initialization seems more gentle.

Overall, we have a working transcoder and I am exceptionally happy :)

Thanks
Adrian

MaikStohn, 2016-04-23:
Hello Adrian,

good to hear it's working.

Some questions / notes to your comments:

> Nozzle height used in transcoder is 233.71 (is 208.12 in UP Software)

Are you sure it's so much off? Or should it read 213.71?

>Extrusion Multiplier = 0.25

This looks like a complete wrong value. Usually extrusion multiplier should be something around 1.00
Can it be that you set wrong filament diameter (Other->Filament properties)?

>Tools=>Options=>Machine=>Build Volume=>X-Axis = 255.00
Tools=>Options=>Machine=>Build Volume=>Y-Axis = 205.00
Tools=>Options=>Machine=>Build Volume=>Z-Axis = 205.00

According to machine's self report it can be: 260,210,210 (X/Y/Z) or 262,213,213 if you want to push to the limits

>It does seem a bit slow to prepare for printing.
Heating the bed from cold takes 10+ minutes, which is not normal when using the UP Software.
Sometimes, it heats the bed but doesn't do anything else. I haven't quite worked out the repeatable pattern here.
Sometimes it seems to wait for the nozzle to cool to less than 45 degrees, before heating it up to temperature again when preparing to print.

For this we should open a new pull request.

>At the end of a print it just stops and does not lower the bed, but I'll fix this with some custom gcode script.

Looks like you removed/disabled standard Simplify3D "Ending Script" (Scripts->Ending Script)
It should read:
```
M104 S0 ; turn off extruder
M140 S0 ; turn off bed
G28 ; home all axes
M84 ; disable motors
```
Which will do exactly what you expect.

>The Initialization procedure works fine, but is a bit more brutal than the UP Software. It has quite an audible clunk as it hits the end stops on the X and Y axis (Z seems fine).The UP Software Initialization seems more gentle.

I will lower the fast homing speed for UP-Box. 
Can you suggest a reduction? Now it is 50mm/sec.

>Overall, we have a working transcoder and I am exceptionally happy :) 

Great. This also makes me happy :-)

Maik

adrianbear, 2016-04-24:
Hi Maik,
Yep, I screwed up. The filament diameter was set to 0.75 instead of 1.75. Not sure how that happened.
The nozzle height is correct though.
Is it prefer that I edit my existing post, or should I post a new set of settings once I have re-calibrated?

Thanks
Adrian
PS. I'd like to make a small donation for the effort you've put into this, which is something I've never done before. Is there an easy way to do that? Could you email me some details so we don't post them publicly or something?

adrianbear, 2016-04-24:
Hi Maik,

Also the Initialization speeds for X and Y I think could be reduced to about 60 % of their current speeds (maybe 30 mm/sec would be good).
I have taken two videos to compare for you available here => [https://we.tl/q584rYFqMh]

Thanks
Adrian

MaikStohn, 2016-04-24:
Hello Adrian,

Next version will have reduced speed for homing.

Donation is absolutely not needed. Just some happy users of this software is enough for me.
If you want to support the project I might ask you to do something a little bit more complicated (for proper UP-Box support).
In case we can not find out why up box is having this startup hick ups the only option will be to use the normal up software, slice a small test model and print it. Then the UP Box must be opened and the micro SD card must be taken and read on a PC. This will give us access to the original ROM-Program-0 which is used for initialization (it is written to SD card for every job by the printer itself).
 
Maik

BTW: I think it's time to close this issue. New features / problems should have their own issues for better tracking.

Gitty-Hub, 2016-04-25:
I can confirm the height issue, my Box was printing in the air for the first test.. But nice to see the options - thanks!

MaikStohn, 2016-04-25:
Regarding the "height" on UP-Box,

Since there is no magic or miscalculation in up3dtranscode it just looks like your printer is having 233mm Z instead of the announced 205mm (maybe this is something to be happy about).

It could be that Tiertime decided to cap 30mm as a safety distance from a (maybe whacky) Z limit switch.
The video from Adrian shows that transcoder Z homing has save distance from limit switch (you can hear the limit switch being triggered and releases properly).

Maik

Gitty-Hub, 2016-04-25:
Yes that is probably right, I was going to post that the initialize of the gcode does not put the Z table back up a bit. Up software initializes and then moves up the table, I just never knew this is extra height on top of the 209 I have with my machine :-)

MaikStohn, 2016-04-26:
I pushed a new version (v0.4.0) which includes slow down for homing for UP-BOX. The new version has big improvements for accuracy and fixes some wrong acceleration calculation.
It also changed the bed heating procedure (wait is a factor to temperature now):

! Do not set bed temperature multiple times. This is will cause long wait times !

Procedure is like this now
- print starts
- as soon as g-code is found which tells to heat the bed it starts heating the bed (slow red/blue blinking)
- it waits a fixed amount of time (based on temperature), e.g. 100°C will wait apx. 3 minutes
- when heating of bed is finished next g-code is used
- in case extruder is requested to heat up it will do so (fast red/blue blinking)
- when extruder got temperature it will start printing (normal red/blue blinking)

Please make sure your fixed start g-code does NOT contain any temperature settings and also make sure not to use multiple temperatures for heated bed (now).

Closing this issue since it should be all fixed by now. Feel free to open new issues for reporting problems or suggestions.

closed 2016-04-26 by MaikStohn