MaikStohn/UP3D #14: up3dwifisrv does not connect from OS X UP Studio 0.0.0.10

kscheff, 2016-04-18:
I compiled the up3dwifisrv under OS X and run it in foreground to see debug output.
When connecting from a windows machine (Settings/Printer) running Up Studio I get the following output:
```
Received UDP packet from 192.168.0.15:54834
Search UP...
Received UDP packet from 192.168.0.15:54834
Search UP...
Received UDP packet from 192.168.0.15:54834
Search UP...
Received UDP packet from 192.168.0.15:54834
Search UP...
Received UDP packet from 192.168.0.15:54833
Connect UP...
```

Running the same procedure from OS X Version of Up Studio I the final `Connect UP...` message is missing.

MaikStohn, 2016-04-19:
Hello,

up3dwifisrv is under development. In development branch is a much newer version.

I just tested it and it works ok with Mac Up Studio. In case it can not connect (gives error message in up studio) make sure your firewall ist not blocking incoming UDP *AND* TCP request for upstudio.app

Maik

MaikStohn, 2016-04-20:
Looks like the current MAC version of UP Studio is "work in progress". Many WiFi related functions are just having code like this inside:

```
int SearchAP(char (*ap)[32])
{
  puts("Please add code for SearchAP()");
  return 0;
} 

```

So it seems just a bit to early to look for bugs in up3dwifisrv.

MaikStohn, 2016-04-29:
Looks like I found the bug... but it is inside of UP Studio! So nothing we have to fix. They send CMD 0x200C for connect (instead of 0x1002 like in Windows version). Since CMD 0x200C is a command intended for setting up the ServerAddress it looks like they just made a copy/paste error.

closed 2016-04-29 by MaikStohn