MaikStohn/UP3D #17: Enhancement Idea

adrianbear, 2016-04-30:
Hi Maik,

Not sure if I should add this as an issue, but I don't see where else to write it.

Do you think we could wrap the up3dtranscode, upload and upshell applications up into a virtual USB/COM gcode port?
i.e. the idea being that Simplify3d (and others like Cura) could just communicate directly rather than dealing with gcode files and manually executing from the command line?
Would be great if the print progress and temperature values could be reported back into Simplify3d during printing.
Or maybe you've already though of this? I am guessing you probably have.

Thanks
Adrian

MaikStohn, 2016-04-30:
Hi,

since normal RepRap operation (which is what Simplify or other tools can handle) is very different from UP this always would be a bad working vehicle. Reprap transfers data sequentially while printing while UP transfer all data first to SD card.

I think it's better to do a nice looking GUI for UP which can do all this (take GCode, transcode, transfer to machine and monitor the process). This will indeed consume less time than to make a virtual RepRap COM port emulation.

Maik

adrianbear, 2016-05-01:
Hi Maik,

I previously had an XYZ Davinci printer I used with Simplify3d and it used GCode via a COM port and sent the job to the printer which in turn stored it on its onboard SDCard. SImplify3d was then still able to monitor the job, temperatures etc and cancel it mid stream. Of course it couldn't change the speeds whilst printing though.

Although I agree it would be easier, I'm not sure why you'd want to create a new GUI when Cura and Simplify3d (among others) already have working GUI's. Some even have web services and network printing capabilities built in for gcode printers (i.e. Repetier Server and Host) that could be leveraged.

Thanks
Adrian

MaikStohn, 2016-05-01:
HI Adrian,

just because UP is not a G-Code printer. In the end we spend more time emulating a G-Code printer to make proprietary tools like Simplify (non free) or Cura (tied to Ultimaker, getting worse) work.

Why not make a more advanced tool completely open source instead (just my thinking)?

kscheff, 2016-05-02:
@adrianbear I use Slic3er and here I can specify an post processing script. I have placed a script to invoke up3dtranscode and upload automatically when exporting to g-code. To support other slicers  I have done a AppleScript version which can observe an folder (folder action) and automatically asks to transcode and upload the data to the printer. Both versions are only available for Mac. I have not updated my scripts to the newest version of the up box, but if you are interested this can be easily done.

adrianbear, 2016-05-03:
From what I can tell GCode is now basically the "standard" for 3d printers. I seriously doubt UP will be able to maintain their own proprietary code much longer anyway, and I certainly doubt they will win out over the plethora of other printers using GCode.
Therefore any new GUI you build for UPCode, will only ever be useful for UP Printers (and only those built in the short to medium term).

I see this tool (the uptranscoder) as valuable because I can use the already existing, already powerful slicers that have already been built. The transcoder will prove to Tiertime, that their software basically sucks, but their hardware is good AND they should focus on what they are good at.

@kscheff I might have a play with Slic3r, but for the moment I am focused on printing RC Aircraft from 3dlabprint and they have a detailed guide for using either Simplify3d (which they recommend for making flying aircraft) or Cura. As you might guess, the STL's won't print from the UP Software or at least not anywhere near satisfactorily.......................................................................................................................................................................................................................................................................................................................................................................
Plus, and I hope I don't get you guys off side here :), I am a dedicated windows guy. I don't know where to start with a Mac.

adrianbear, 2016-05-03:
Hmm, it looks like Repetier provides exactly the integration scenarios I am talking about and they were Open Source, but not anymore. :(

[Repetier API](https://www.repetier-server.com/manuals/programming/API/index.html)
[Goes Closed Source](http://forums.reprap.org/read.php?267,313864)