MaikStohn/UP3D #20: Question / Issue regarding multiple UP machines attached to a single computer

adrianbear, 2016-05-01:
Hi Maik,
I currently only have 1 printer, but now that we pretty much have this transcoder working, I am considering buying a second UPBox.
How does the upload.exe know which UPBox to send the umc file to?
I suspect this a setting that should be part of the upload.exe
Also which printer would the upshell read data from? I suspect this application would need a setting for it too.
I noticed the UP Software seems to differentiate between multiple printers by a serial number.
Normally with gcode printers they are differentiated by the COM port they are attached to.

Thanks
Adrian

MaikStohn, 2016-05-01:
Hello Adrian,

the current version (command line) is for testing / development purposes only.

A nice GUI based version is in the works which will have support for multiple printer (USB and WiFi), 3D preview, nice print monitoring and even pause of print.

Maik

kscheff, 2016-05-02:
@MaikStohn is there another repository for this already? What frameworks are you planning to support.

MaikStohn, 2016-05-02:
Hi kscheff,

I'm still evaluating / experimenting. For sure it will use QT5.x + OpenGL2. So nothing public yet.

Maik

balex3, 2016-06-01:
Hi Maik,

The USB PID for my UP Plus is 277D instead of 2777
So the development software cannot connect to the printer.
Is this easily changed?

Thanks,
Alex