MaikStohn/UP3D #21: Question: G-Code 'flavour'

companje, 2016-05-03:
Thanks for your great effort!

I was wondering: When creating GCODE from Cura, does it matter for which printer is selected? And therefore which GCODE flavour is used?

When I select `Ultimaker 2go` in Cura `up3dtranscode` shows this error when executed: `GCP ERROR at line 11: UNSUPPORTED command >>G10`. So added a 'Prusa i3' in Cura. Now the `output.umc` file is succesfully created.

What printer / slicer settings should I use in Cura?

dfyx, 2016-05-03:
Related to #15

MaikStohn, 2016-05-03:
Hi,

I implemented most standard G-Codes which is compatible with flavor "RepRap" or "Marlin" or ...

In case you find G-Codes flavors which you think should be supported and get an error, then please report them (like you did). I will then have a look if it can be supported / fixed.

BTW: G10 is for TOOL-OFFSET (e.g. when using multiple nozzles) which I doubt we ever need for our old UP printer.

Maik

companje, 2016-05-03:
Below are the first lines of the gcode generated by Cura when the Ultimaker 2go is selected. It's UltiGCode which results in that it has (almost) no startgcode for homing and also no temperature info since that's configured on the printer itself.

I'm ok with selecting a RepRap flavor in Cura like the Prusa i3. That will do it for now.

```
;FLAVOR:UltiGCode
;TIME:719
;MATERIAL:705
;MATERIAL2:0
;NOZZLE_DIAMETER:0.4

;Generated with Cura_SteamEngine 2.1.0-
;LAYER_COUNT:98
;LAYER:0
M107
G10
G0 F7200.000000 X49.623 Y49.349 Z0.300
;TYPE:SKIRT
G11
G1 F1800.000 X50.222 Y48.834 E0.09479
G1 X50.870 Y48.383 E0.18953
G1 X51.561 Y47.999 E0.28440
G1 X52.286 Y47.687 E0.37911
...
```

MaikStohn, 2016-05-04:
Hi,

Since a gcode without temp settings is not gonna work at all with UP (or any other RepRap printer) I think we have nothing to do here. Just need to tell our users to use normal (RepRap/Marlin G-Code style).

Maik

closed 2016-05-04 by MaikStohn