MaikStohn/UP3D #22: UPload stops at 30%

companje, 2016-05-03:
When uploading my `cube.umc` (111kb) file from OSX right after turning to the UPMini it gets stuck at 30%. The power led is purple (alternating between red and blue very fast). 

```
./upload cube.umc
UPload: 32.34 kB sent (30%) [11.87 kB/sec]
```

Any clues?

MaikStohn, 2016-05-03:
Hi,

can you run "upinfo" please and provide the output of it? Maybe another corner case with an "older" firmware version.

Thanks,

Maik

companje, 2016-05-03:
Here's the result of 'upinfo`
```
UP 3D PRINTER INFO:
TypeID:2713 Serial:71067 ROM:6.0600 Model:UP Mini U1:1.00 U3:1 U4:102 U7:100
Max-X:-120.000000 Max-Y:120.000000 Max-Z:130.000000 Steps/mm-X:854.000000 Steps/mm-Y:854.000000 Steps/mm-Z:854.000000 Steps--A: 40.000000 Print sets: 4
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 1"
  ND:0.30 LT:0.20 SW:0.50 ST:1.00 HW:0.52 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:35.00
  HV:45.00 XV:50.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.00
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#1  Name:"Set 2"
  ND:0.30 LT:0.25 SW:0.53 ST:1.00 HW:0.55 HS:6.00 HL:3.00 XW:0.57 XS:8.00 XL:3.00 SV:40.00
  HV:50.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.33
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#2  Name:"Set 3"
  ND:0.30 LT:0.30 SW:0.55 ST:1.00 HW:0.57 HS:6.00 HL:3.00 XW:0.60 XS:8.00 XL:3.00 SV:40.00
  HV:50.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.65
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#3  Name:"Set 4"
  ND:0.30 LT:0.35 SW:0.60 ST:1.00 HW:0.63 HS:6.00 HL:3.00 XW:0.66 XS:8.00 XL:3.00 SV:40.00
  HV:50.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.10
  O1:20.00 O2:6.00 O3:1.00 O4:2000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
```

companje, 2016-05-04:
Is it possible to update the firmware on my old UPMini?

MaikStohn, 2016-05-04:
Hi, firmware update is not possible for all UP printers.

You only can update the so called ROM (which is just data, same as what upinfo is showing). Tiertime has not implemented a firmware update procedure or they are afraid that someone could "steal" the firmware and make copy printer.

Could you open the original UP software please and provide the "Model / Rom / Firmware" versions displayed there ("Help->About UP...").

Maik

companje, 2016-05-04:
When I run UP! V2.18 on Windows the Help->About UP... dialog shows:
```
Model:      Up Mini
Serial ID:  71067
Rom:        V6.060
Firmware:   V3.04
```

MaikStohn, 2016-05-04:
Ok, this means it is covered by the old firmware work around already.
This means for some reason your UP reports 0 free memory blocks after 32.4kB written to SD.

Next thing is to turn on debug output for usb communication. In case you can compile it yourself you have to uncomment the following line in "common/up3dcomm.c": 
```
//#define _DEBUG_IN_OUT_
```
change to:
```
#define _DEBUG_IN_OUT_
```

If you can't compile yourself I will add a special version in next release.

Maik

companje, 2016-05-06:
Here's some of the debug output. It appears to me that when it get's stuck it keeps receiving 46 and keeps sending 00 00 00 00 06 as a respons.

```
$ ./upload /Users/rick/Downloads/UP3DTOOLS/cube.umc
>01 00
<30 01 00 00 06
>56 8E 0A 00 00 00
<06
>56 10 13 00 00 00
<06
>43
<06
>6C 05 01
<06
>46
<9E 07 00 00 06
>2F 03 06 00 00 00 0A 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 0B 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 4D 00 00 00 57 03 00 00 00 00 00 00 00 00 00 00
>46
<9B 07 00 00 06
>2F 03 06 00 00 00 4C 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 02 00 00 00 00 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 05 00 00 00 A0 0F 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>46
<98 07 00 00 06
>2F 03 06 00 00 00 41 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 42 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 43 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>46
<95 07 00 00 06
>2F 03 06 00 00 00 46 00 00 00 0D 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 49 00 00 00 50 00 00 00 00 00 00 00 00 00 00 00 08 00 00 00 00 00 00 00 00 00 48 42 00 00 80 C0 00 00 00 00
>46
<92 07 00 00 06
>2F 03 08 00 00 00 00 00 00 00 00 00 20 41 00 00 00 C0 00 00 00 00 08 00 00 00 01 00 00 00 00 00 48 42 00 00 80 40 00 00 00 00 08 00 00 00 01 00 00 00 00 00 20 41 00 00 10 41 00 00 00 00
>46
<8F 07 00 00 06
>2F 03 08 00 00 00 02 00 00 00 00 00 48 42 00 00 C0 C0 00 00 00 00 08 00 00 00 02 00 00 00 00 00 40 40 00 00 00 C0 00 00 00 00 06 00 00 00 10 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
>46
<8C 07 00 00 06
>2F 03 06 00 00 00 11 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 35 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 17 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
>46
<89 07 00 00 06
>2F 03 06 00 00 00 82 00 00 00 FF 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 83 00 00 00 FF 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 43 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00
>46
<86 07 00 00 06
>2F 03 06 00 00 00 3B 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 16 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 8E 00 00 00 90 01 00 00 00 00 00 00 00 00 00 00
>46
<83 07 00 00 06
>2F 03 05 00 00 00 E0 93 04 00 00 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 8E 00 00 00 C8 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 4D 00 00 00 1E 02 00 00 00 00 00 00 00 00 00 00
>46
<80 07 00 00 06
>2F 03 06 00 00 00 4C 00 00 00 24 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 17 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 41 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00
>46
<7D 07 00 00 06
>2F 03 06 00 00 00 39 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 14 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 17 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
>46
<7A 07 00 00 06
>2F 03 06 00 00 00 41 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 39 00 00 00 FA 00 00 00 00 00 00 00 00 00 00 00 06 00 00 00 14 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
>46
<77 07 00 00 06
>2F 03 06 00 00 00 8E 00 00 00 64 00 00 00 00 00 00 00 00 00 00 00 07 00 00 00 12 00 00 00 01 00 00 00 3D 00 00 00 00 00 00 00 06 00 00 00 8E 00 00 00 C8 00 00 00 00 00 00 00 00 00 00 00
>46
<74 07 00 00 06
>2F 03 06 00 00 00 4D 00 00 00 A6 01 00 00 00 00 00 00 00 00 00 00 06 00 00 00 4C 00 00 00 32 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 00 40 1C C6 00 00 00 00 00 40 1C C6 00 00 00 00
>46
<71 07 00 00 06
>2F 03 03 00 00 00 00 00 00 80 00 00 00 00 00 00 00 00 00 00 00 00 03 00 00 00 00 00 00 80 00 00 00 00 00 00 00 80 00 00 00 00 03 00 00 00 00 40 1C C6 00 00 00 00 00 00 00 00 00 00 00 00
>46
<6E 07 00 00 06
>2F 03 03 00 00 00 00 A0 0C C6 00 00 00 80 00 A0 0C C6 00 00 00 00 03 00 00 00 00 A0 0C C6 33 33 D8 C2 00 A0 0C 46 00 00 00 00 06 00 00 00 0A 00 00 00 01 00 00 00 00 00 00 00 00 00 00 00
>46
<6B 07 00 00 06
>2F 03 06 00 00 00 0B 00 00 00 00 00 70 41 00 00 00 00 00 00 00 00 06 00 00 00 4D 00 00 00 9C 01 00 00 00 00 00 00 00 00 00 00 06 00 00 00 4C 00 00 00 33 00 00 00 00 00 00 00 00 00 00 00
>46
<68 07 00 00 06
[... 1929 lines cut ...]
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
>46
<00 00 00 00 06
^C
```

MaikStohn, 2016-05-06:
Hi,

good that you can compile it yourself :-)

```
>46
```
is the command sent to printer to "GetFreeBlocks in RAM"
```
<XX XX XX XX 06
```
is the answer from printer (06 at end of every response means command response "OK")

So at the beginning it reports 9E 07 00 00 (reversed byte order) ==> 0x0000079E = 1950 free RAM blocks - this is ok.

Then upload sends some data blocks to printer  ">2F..." and the reported GetFreeBlocks in RAM decrease (normal behavior).

Usually the printer will write the data to SD card as soon as enough blocks are in RAM buffer
(1 block = 20 byte, write to SD card waits for >100 blocks (2kB) in RAM buffer for efficient SD card writes). 

However your log shows that the RAM buffer fills up until 0xFB blocks are free (==> 1699 block written = 33980 bytes) maybe earlier or at least then it should write some data to SD card and report a bigger GetFreeBlocks in RAM after the flush.
Instead it always reports 0 free blocks (which let's upload wait until the data is flushed and written to SD - which just does not happen).

So things to try for you:
1. Try to change the program number to 9 (up original software always writes to program slot 9, I just used another one)
==> in "upload.c":
```
uint8_t program_id = 5;
```
change to:
```
uint8_t program_id = 9;
```

2. For testing we should manually set "write to sd card" state
==>  in "upload.c"

after the line
```
UP3D_SetParameter(PARA_PRINT_STATUS,0x13); //write to SD
```
add
```
UP3D_SetParameter(0x37,1);
```

Maik

companje, 2016-05-06:
Changing the program_id to 9 and adding `UP3D_SetParameter(0x37,1);` and then recompiling again unfortunately did not change the behaviour.

I notice something else. Just after 3% it sends multiple 46's without getting a response. Is that ok?
```
>46
>46
>46
>46
<9F 06 00 00 06
```

companje, 2016-05-06:
I found the problem. It was the SD-Card causing trouble. I tried to print something with the official UP software and that did not work (anymore) either. I opened the machine placed a fresh 8GB SD-card and now I'm able to upload the full .UMC file through your upload tool.

Thanks for your help! I'm closing this issue now.

closed 2016-05-06 by companje