MaikStohn/UP3D #23: Rename tools to be less generic

dfyx, 2016-05-03:
For users who want to put the tools somewhere in their `PATH`, it would be good, if some of the tools were renamed. Especially `upload` is pretty generic and could collide with other tools or custom scripts (for example I have a script named `upload` that uploads files to a public directory on my web server).

For consistency you might start all names with `up3d` or `up3d-` so `upload` might become `up3dload`, `up3d-load`, `up3dupload`, `up3d-upload` or something similar. I think I prefer the last one (which would make the other tools `up3d-transcode` and `up3d-shell`)

MaikStohn, 2016-05-03:
You are right,

will do that soon.

Maik

MaikStohn, 2016-05-06:
Hi, 

new version is out, Tools renamed to always have "up3d" in front.

Maik

closed 2016-05-06 by MaikStohn