MaikStohn/UP3D #24: Question: Is there a maximum temperature

adrianbear, 2016-05-06:
Hi Maik,

Now that I can use other slic3rs..Are there a maximum temperatures I should keep under for the extruder and the bed (specifically on my UP Box)?

Thanks
Adrian

MaikStohn, 2016-05-07:
Hi,

a heater is mostly limited by the power provided to it. The power supply, the electronics in up and the connection cables are not big enough to generate "to high temperatures" at all.
So even if you set bed temperature to 1000°C it will never reach it. The heated bed in UP-Mini for instance can not get hotter than 95°C at all. 

The extruder heater can produce higher temperatures (>300°C) which might damage the thermistor (used to check temperature). This we just don't know yet since it's unknown what kind of thermistor Tiertime used. 

So using <300°C for extrusion temperature should be safe.

Maik

adrianbear, 2016-05-15:
Excellent, thanks @MaikStohn

closed 2016-05-16 by MaikStohn