MaikStohn/UP3D #25: Question / Issue : convg does not produce sensible gcode in preview

adrianbear, 2016-05-07:
Hi,
I've just compiled up the convg tool and run it over the MemDump.dat file from FixUp3d.
The MemDump.dat was generated from a simple 10x20 cylinder print.
I then previewed the resulting gcode in Simplify3d. The preview shows the print head going all over the place.
[image]

I also ran the convg tool over the upcode generated file from up3dtranscode. Ie from gcode to upcode and back to gcode again.
The preview is even worse again.
Below is a preview of the source gcode
[image]

After transcoding and then converting back to gcode again it looks like this
[image]

I understand that the transcoder "mostly" works, so why can we convert back and forth and get the same results?

Thanks
Adrian

MaikStohn, 2016-05-07:
Hi,

convg is a small test tool. It has hard coded the steps/mm for UP-Mini (854.0).

When you use it with other machine codes (e.g. UP-Box) you must change the steps/mm.

E.g. for UP-Box change:
```
#define STEPS_X (854.0)
#define STEPS_Y (854.0)
#define STEPS_E (854.0)
```
to
```
#define STEPS_X (644.0)
#define STEPS_Y (644.0)
#define STEPS_E (854.0)
```
STEPS E is same for both types.

Then output will be correct for your machine program.

Maik

adrianbear, 2016-05-07:
Hi Maik,

Right you are :) All good now I've adjusted the X Y step constants in the programs.

Thanks
Adrian

closed 2016-05-07 by adrianbear