MaikStohn/UP3D #26: UP!Mini stops after having printed some layers

companje, 2016-05-07:
I'm currently investigating a problem where my UP!Mini stops after having printed some layers. In my second attempt it stopped at the same place with the same UMC file. I'm trying with another UMC file now.

This was the state in UP!3D Shell when it stopped.
```
                                                     UP!3D SHELL

Machine-State: (2) Running Program     Program-State: (0) Program Stop      System-State: (01) System Ready

Motor 1: State: (0) STOP      | Limit+: (00000000) | Limit-: (00000000) | Axis X State: (0) No Error    Pos: -52948
Motor 2: State: (0) STOP      | Limit+: (00000000) | Limit-: (00000000) | Axis Y State: (0) No Error    Pos: 56364
Motor 3: State: (0) STOP      | Limit+: (00000000) | Limit-: (00000000) | Axis Z State: (0) No Error    Pos: -105896
Motor 4: State: (0) STOP      | Limit+: (00000002) | Limit-: (00000000) | Axis A State: (0) No Error    Pos: 181040

Print-State: Not Printing Layer: 1    Height: 2.470 Progress:  30%                            Time Remaining:  0:17:46

Heater 1: ActualTemp: 240.2 C TargetTemp: 240.0 C SetTemp: 240.0 C Temp Reached: YES Hea *** HF 1: 0
Heater 2: ActualTemp: 248.1 C TargetTemp: 230.0 C SetTemp:   0.0 C Temp Reached: YES Hea *** HF 2: 0
Heater 3: ActualTemp:  51.4 C TargetTemp:  50.0 C SetTemp:  50.0 C                   Hea *** HF 3: 0
Heater 4: ActualTemp: 113.1 C TargetTemp: 105.0 C SetTemp:   0.0 C                       *** HF 4: 32

Have Support: No  Feed Error Length: 2000 Feed Back Length:   0.000 Change Nozzle Time:   0.000 Jump Time:   0.000 Usi

Check Door:    1 Check 24V:    0 Check PowerKey:    1 Check LightKey:    1 Check WorkRoomFan:    0

Read From SD Card: 0 Write To SD Card: 0 PausePRG: 0 StopPRG: 0 InitPRG: 0 ISTATE: 0

3D:00000000 3E:00000130 3F:00000000 40:00000008 41:000000f0 42:00000000 43:00000032 44:00000000 45:00000000 46:0000000
48:00000000 49:00000050 4A:00000000 4B:00000000 4C:0000001e 4D:0000043c 4E:00000000 4F:00000000 50:00000000 51:0000000
54:00000000 55:00000000 56:00000000 57:00000000 58:00000000 59:00000000 5A:00000000 5B:00000000 5C:00000000 5D:0000000
60:00000000 61:00000000 62:00000000 63:00000000 64:00000000 65:00000000 66:00000000 67:00000000 68:00000000 69:0000000
6C:00000000 6D:00000000 6E:00000000 6F:00000000 70:00000000 71:00000000 72:00000000 73:00000000 74:00000000 75:0000000
78:00000000 79:00000000 7A:00000000 7B:00000000 7C:00000000 7D:00000000 7E:00000000 7F:00000000 80:00000000 81:0002c35
84:bc03147b 85:3ed84091 86:00000000 87:00000000 88:43d48000 89:449b0000 8A:00000000 8B:00000000 8C:00000001 8D:0000000
90:00000000 91:00000000 92:00000000 93:00000002 94:00000000 95:00000001 96:00000000 97:3eb851ec 98:3eb851ec 99:3d75c28
9C:c1200000 9D:c1200000 9E:c1200000 9F:00000000 A0:00000000 A1:00000000 A2:00000000 A3:00000000 A4:00000000 A5:0000000
A8:00000000 A9:00000000 AA:00000000 AB:00000000 AC:00000000 AD:00000000 AE:00000000 AF:00000000 B0:00000000 B1:0000000
B4:00000000 B5:00000000 B6:00000000 B7:00000000 B8:00000000 B9:00000000 BA:00000000 BB:00000000 BC:00000000 BD:0000000
C0:00000000 C1:00000000 C2:00000000 C3:00000000 C4:00000000 C5:00017fca C6:00017f62 C7:00000003 C8:00000000 C9:0000000
CC:00000000 CD:00000000 CE:00000000 CF:00000000 D0:00000000 D1:00000000 D2:00000000 D3:00000000 D4:00000000 D5:0000000
D8:00000000 D9:00000000 DA:00000000 DB:00000000 DC:00000000 DD:00000000 DE:00000000 DF:00000000 E0:00000000 E1:0000000
E4:00000000 E5:00000000 E6:00000000 E7:00000000 E8:00000000 E9:00000000 EA:00000000 EB:00000000 EC:00000000 ED:0000000
F0:00000000 F1:00000000 F2:00000000 F3:00000000 F4:00000000 F5:00000000 F6:00000000 F7:00000000 F8:00000000 F9:0000000
FC:00000000 FD:00000014 FE:00000010 FF:00000010
84:-0.008000 85:0.422368 86:0.000000 87:0.000000 88:425.000000 89:1240.000000 0
97:0.360000 98:0.360000 99:0.060000 9A:0.360000 9B:-10.000000 9C:-10.000000 9D:-10.000000 9E:-10.000000
```

MaikStohn, 2016-05-07:
Hi,

in case you can please attach the UMC file for inspection. This kind of problem really should be investigated and solved.

companje, 2016-05-07:
Hi Maik,

This ZIP file contains both the GCODE and the UMC file: [bunny-3cm-fails-on-upmini.zip](https://github.com/MaikStohn/UP3D/files/253550/bunny-3cm-fails-on-upmini.zip).

I managed to print another smaller version of the same model without problems. So I guess there's something wrong in my GCODE or in the UMC file.

An interesting fact is that my third attempt with the same file it stopped again as suspected but somehow the heating of the nozzle lost control. The nozzle became so extremely hot (I guess around 300 degrees) that a lot of black smoke came out of the printer...

MaikStohn, 2016-05-08:
Hi,

from your description it sounds like the UP internal CPU crashed which let it fail to control temperature.

Most likely this was caused due to a bad machine command read from SD card. I will investigate the UMC file you provided but it also might been caused by a bad SD card read.

You reported in another issue that you had SD card problems and changed it. Here some suggestions:
- make sure the SD card is secured properly in your UP printer. It might get loose in the standard holder due to machine vibrations. I fixed mine with a small (scotch) tape which ensures it is always inserted properly and can not shake
- take out the SD card and completely erase it (e.g. use a tool like "dd" to write all sectors with 0)

In case you reproduce the failure reliable at the exact same position in print then it would be very helpful if you can take out the SD card after a failed print and read the complete content, ZIP it and provide it.
We then could check if SD card content written is exact same as what we wanted to write.

Maik

companje, 2016-05-09:
I will upload the UMC file to another SD card to make sure it's not the card. Then after the crash I will provide you with a zipped image of the SD-card.

Virtualmk, 2016-06-06:
Hi, I have the same problem with my mini. Different models, different slicers, formatted SD card, result is the same - nozzle stops after printing few layers. 
My test cube GCODE with the UMC [myobject.zip](https://github.com/MaikStohn/UP3D/files/301211/myobject.zip)

ZonSoft, 2016-06-22:
The same problem here. Stops right during the first layer print, reproducable with different models. Here is the sample UMC.
[output.zip](https://github.com/MaikStohn/UP3D/files/327660/output.zip)

DerekL, 2016-06-23:
Same issue here for an Up Plus! 
Multiple different models on multiple different attempts: print freezes after first few layers.

MaikStohn, 2016-06-27:
Hi,

I will have a look on it. Thanks for the files.

TyberiusPrime, 2016-08-02:
I can also offer a pretty stripped down gcode (and umc) file - got it down to 300 odd lines, with the second to last (functional) line triggering the error.
[invader_minimal_bug.zip](https://github.com/MaikStohn/UP3D/files/397019/invader_minimal_bug.zip)

TyberiusPrime, 2016-08-02:
Bisection testing shows me that I can print with release 0.5.0, but I can't print with release 0.6.0.

Gotta get the compilation set up to pinpoint the exact commit...

TyberiusPrime, 2016-08-02:
ok, so 63e5b3cf13dfce4526748cccc99721dc73979f33  fails, and 9ac3112eed854e6d06af2ad223cddc4f566c1643  works.

I fail to see how the reduced acceleration settings create the issue though?!?

dfyx, 2016-08-02:
Might be that there's a rounding error that generates an acceleration of 0 or something like that.

TyberiusPrime, 2016-08-02:
Possibly. I can now confirm, patching the current master with acceleration = 2000, 2000 leads to a working umc. Please find attached the invader_minimal gcode from above in both ok (acceleration=2000) and broken (1500).

[invader_umc_contrast.zip](https://github.com/MaikStohn/UP3D/files/397325/invader_umc_contrast.zip)

TyberiusPrime, 2016-08-02:
> Might be that there's a rounding error that generates an acceleration of 0 or something like that.

sa_x, sa_y and sa_a (line 175, hoststepper.c) are sometimes 0 with both 2000 and 1500 acceleration for the invader_minimal gcode.

TyberiusPrime, 2016-08-03:
I instrumented the umc writer and it's helper.
At acceleration = 1500, this is a diff of the result of Invader.gcode with the line triggiering the bug and without:

```
377d376
< planner_add  X=43.017000 Y=40.686000 A=2.406210 F=1200.000000
1043,1048c1042,1044
< UP3D_PROG_BLK_MoveL p1=54, p2=64448, p3=0, p4=0, p5=3403, p6=0, p7=0, p8=217
< UP3D_PROG_BLK_MoveL p1=1, p2=8706, p3=512, p4=0, p5=2048, p6=0, p7=0, p8=0
< UP3D_PROG_BLK_MoveL p1=32, p2=64460, p3=0, p4=0, p5=15152, p6=0, p7=0, p8=-218
< UP3D_PROG_BLK_MoveL p1=3, p2=60631, p3=0, p4=-7509, p5=512, p6=0, p7=-910, p8=0
< UP3D_PROG_BLK_MoveL p1=172, p2=65423, p3=-17, p4=-11442, p5=994, p6=0, p7=0, p8=0
< UP3D_PROG_BLK_MoveL p1=11, p2=60380, p3=0, p4=-10518, p5=884, p6=0, p7=956, p8=-76
---
> UP3D_PROG_BLK_MoveL p1=48, p2=65430, p3=0, p4=0, p5=3444, p6=0, p7=0, p8=224
> UP3D_PROG_BLK_MoveL p1=1, p2=7064, p3=512, p4=0, p5=1536, p6=0, p7=0, p8=0
> UP3D_PROG_BLK_MoveL p1=64, p2=64744, p3=0, p4=0, p5=14048, p6=0, p7=0, p8=-219
mh /martha/temp/0.6.1/UP3D/UP3DTRANSCODE $ 

``` 

Does this suggests anything to any of you devs?

TyberiusPrime, 2016-08-03:
Disregard my earlier analysis with regard to 'it didn't happen with versions <=0.5' - I have a (substantial) gcode here that shows the same stop in earlier layers :(

TyberiusPrime, 2016-08-10:
Any ideas what I could try next?

link0007, 2016-09-25:
I just tried using the G-code transcoder for the first time. I sliced a model in slic3r, transcoded it and loaded it.

The entire machine froze after a dozen or so layers. I tried again, and it froze at the same exact moment.

Is there a way to fix this problem? Maybe even just a temporary hack to prevent it from happening? I'll even settle for a way to know whether or not a job is going to crash the machine.

I really don't want to have the uncertainty of big prints failing because of a machine crash.

TyberiusPrime, 2016-10-05:
@link0007 you can always try a test print without any filament to see if it finishes :(.

arhi, 2016-10-05:
using 500R pot hooked to hotend sensor connector instead of pt100 (disconnect hotend completely) dialed around 200R and you don't even need to waste energy heating the hotend for the test print :D

kscheff, 2016-10-05:
@link0007 @TyberiusPrime have you tried with the new version? This version changes the calculation of the movement parameters and might influence the crash you observe.

link0007, 2016-10-13:
@kscheff It still crashes on pretty much every file I throw at it. I am generating the gcode with slic3r. Do you think a different slicer would solve the issue?

kscheff, 2016-10-13:
Which Printer, what Version? STL source, Gcode File and generated UMC file with all steps involved might be helpful to understand what is going on.

mourrr, 2016-11-23:
Not sure if this thread is still active, but I am having the same issue as well. The print stops after a certain number of layers at the same spot. I can replicate this whether the g-code has been generated by Slic3r or Simply3D. changed few parameters such as speed, acceleration, extrusion speed...no avail. 
I am using an up! Plus on OSX with the latest Gui (0.0.34) of the transcoder. The STL was generated from solidworks and prints no problem with the up software .

lovelytwo, 2017-01-20:
I believe I have the same problem as other users have reported in this bug report. That a particular gcode file will consistently stop at the same point.

Background:
-	UP MINI with FW3.30
-	Slic3r 1.29
-	Latest Version from UP3D git repository
-	Compiled on OSX 10.12

My initial experimentation found that both the feed rates and accelerations would change the position at which the print would stop. As other users, have reported.

After two weeks of futile experimentation including using usbpcap with original UP software, modify up3dtranscode to generating very similar UMC (2 > p1 > 500 and 12500 > p2 > 25000 in steps of 50) with only the same machine parameters being set in the same order as the UP software does. Which all seemed to amount to nothing :(

In a move caused by complete frustration, I added periodic umcwriter_pause instructions every 50 gcode lines, and this seem to solve my problem. I then reverted my modifcations except for the periodic umcwriter_pauses and I managed to do three test prints with different feeds and accelerations.

For further validation, I tried printing myobject from this thread. Which stops around layer 5, during the high speed internal fill printing. It stopped without and with pauses. Additionally, I tried preventing up3dtranscode from creating segments with p1 less than 2, and p2 periods less than 12500 to no avail. However, with both the pauses and highly modified up3dtranscode (only the hoststepper mods, p1/p2 emulating the up software) it prints without issues.

I would release my sources, but I currently have a bug. Which requires disabling of the global error correction code. In the mean time if you are having this problem, try printing slower with maximum feeds of 25mm/s and adding “G4 P1” to your slicer software to be performed on layer change, or during retracts.

I will capture some test prints from the UP software at higher speeds to see if and how it’s UMC changes. Then do some more experimentation to determine what our printers have a problem with.

halfluck, 2017-06-24:
@lovelytwo are you able to offer any suggestions or a code snippet on how you implemented he  umcwriter_pause instructions?

Many Thanks Rob.

maszrums, 2020-10-08:
Thank you for your last comment. Can you say me everything about your working softwares (f.ex. software, version and config of your slicer or version of your up3d)? 
Everything works fine in Up Studio, but options are too limited for me.
When I was trying with Up3D v.0.7.3 and standard settings of last Cura it is almost working ok (movements), but filament hardly pushes out and then the extruder drive clogs after a while. Then, on the next print attempt, when I slightly increase the "flow" in Cura, it comes to a situation that the filament first pushes out terribly quickly from the nozzle and then it does not go anymore. 

Please, I need this :)

BiffBish, 2020-12-16:
Sorry to revive a dead thread but I'm having the exact issue stated. I do t fully understood the solution to the problem? I've tried tweaking the overall speed multiplier and junction argument but I can't get it to continue printing. Right now it freezes before the first layer is done so the advice above about adding a G4 P1 command on layer change can't help cause it never gets to a layer change. I see a code snippet above but I don't fully understand how to use it and don't know what it does. Could someone help me out I'll be happy to give files in a bit

Piter-d, 2020-12-17:
Hello,

To help you i join a patched winusb.dll (special thanks to great master MailStohn for original). This patch change 2 things from original : 
- make a file umc/UP.umc (raw file to send to UP mini using up3dload tool (maybe some option can do it without patch))
- force raw file UP.umc to make light on and make bed to 65° during all printed time and nozzle to 200° (PLA for me)
(constant bed temp is very important if you print on glasse bed without rad like me)
[winusb.zip](https://github.com/MaikStohn/UP3D/files/5712041/winusb.zip)

I join too another patch to change temperature (named temp.exe)
to use that, syntax is : 
./temp.exe umc_source_file umc_dest_file nozzle_temp bed_temp
for ABS exemple
./temp.exe UP.umc dest.umc 235 90
[temp.zip](https://github.com/MaikStohn/UP3D/files/5712068/temp.zip)

Shame on me because i'm not a dev man and my code is not smart

Regards

BiffBish, 2020-12-17:
> Hello,
>
>
>
> To help you i join a patched winusb.dll (special thanks to great master MailStohn for original). This patch change 2 things from original :
>
> - make a file umc/UP.umc (raw file to send to [...]

Will this help with the printer stoping randomly? I don't know if your referencing me or not

Piter-d, 2020-12-17:
If i compare two methods : 

Slicer with original UP.exe and patched winusb.dll
- make an umc file (UP.umc) (previous post)
- print this umc file with ./up3dload tool
=> no random stop

Other slicer who make a gcode file
- use gcode generated with ./up3dtranscode tools to make umc file
- print this umc file with ./up3dload tool
- => random stop

I notice a difference between this two ways. ./up3dtranscode use UP3DPCMD_MoveF (0x03) primitive and original UP.exe slicer use UP3DPCMD_MoveL (0x04). Maybe a begining of explanation

Regards

BiffBish, 2020-12-17:
Alright so there's another program that will get me the .umc file you said it's Up.exe? Where could I find that and how exactly does it function. Does it convert gcode to umc or does it take in an .stl or other 3D object format and give the .umc . Thanks for the quick replies!

maszrums, 2020-12-17:
I've found a solution with Prusaslicer and UpStudio2.x with Python script. The only problem was with bed temperature after 1 hour. I cannot test it through 2 weeks, but if you need this still in next year, then I can show this. If so, remind me in January.

Piter-d, 2020-12-17:
@BiffBish 

I don't know what is your tool.

An umc file is a file you can use with ./up3dload tool (from great master MailStohn)

To make an umc file, i know two ways.

1) with original UP Mini tool (mine is v2.13 firmware V3.3) and patched winusb.dll (look at previous post)
=> you normaly print a stl file and a umc/UP.umc file is generate.

2) with a slicer like cuda
=> use cuda and make a gcode file
=> transform gcode file with ./up3dtranscode then a umc file will be make

Regards

Piter-d, 2020-12-17:
> Thank you for answers. In the meantime, I've found a solution with Prusaslicer and UpStudio2.x with Python script. The only problem was with bed temperature after 1 hour. I cannot test it through 2 weeks, but if you need this still in next year, then I can show this. If so, remind me in January.

Happy to know you have a solution. Nice if you can publish this script in 2021 (when you have time because it is not in a hurry for me)

Happy new year

BiffBish, 2020-12-17:
Are you talking about UpStudio? [https://www.tiertime.com/up-studio/](https://www.tiertime.com/up-studio/)

Because that's what I was using previously to print. In that program it just prints immediately to the printer with no "save as umc" option. Is that what your windoesusb.dll thing does?

Piter-d, 2020-12-17:
> Are you talking about UpStudio? https://www.tiertime.com/up-studio/
> 
> Because that's what I was using previously to print. In that program it just prints immediately to the printer with no "save as umc" option. Is that what your windoesusb.dll thing does?

I know Upstudio. I dont use because (maybe it's fixed for now) it hang sometime.
Previous Upstudio it was "Up" https://www.tiertime.com/software/software-archive/ (soft given with my printer).winusb.dll work only with Up for me (i use v2.13)

maszrums, 2020-12-17:
> Are you talking about UpStudio? https://www.tiertime.com/up-studio/
> 
> Because that's what I was using previously to print. In that program it just prints immediately to the printer with no "save as umc" option. Is that what your windoesusb.dll thing does?

He is using Up.exe software. You can download it from:
https://www.tiertime.com/software/software-archive/
(at the bottom of the page)