MaikStohn/UP3D #28: Layer Positioning Issue

adrianbear, 2016-05-10:
Hi Maik,
Testing with my UPBox, there is an issue with the layer positioning.
The error can be seen both in a real print and when converting back to gcode.
This is the original preview
[image]

After converting to up code and then back to gcode, the preview looks like this
[image]

I considered this may just be an issue with the preview only from converting back and forth so I printed it anyway, but the real print has the layers off too.
[image]
[image]

Adrian

MaikStohn, 2016-05-10:
Thanks for the files.

Will have a look at them. The problem seems to be very very small segments (time wise), This only seems to happen when your speed is very slow.

adrianbear, 2016-05-10:
The print shown here was done at 3600 mm/min which is only a bit slower then the UPBox turbo speed  on a 0.4 mm layer height (5020 mm/min).
I also originally tried the print at 4800 mm/min but got the same result.

MaikStohn, 2016-05-11:
I will have a look as soon as I find some time for it.

MaikStohn, 2016-05-12:
Hi,

this was a tricky one. Actually everything was working perfect except a small thing. The absolute Z-move for layer change was not synchronized with the normal movement buffer, which was caused by a skip (ultra small timing segment). So I was looking at wrong place for some time.

New version contains a fix for it. Please check and close the issue if it's solved.

Maik

adrianbear, 2016-05-12:
Yeah, the windows build is taking a while

MaikStohn, 2016-05-13:
Finally the build was getting through. v0.6.4 is ready now.

adrianbear, 2016-05-15:
This appears to be resolved in version 0.6.4. Closing the issue :)
Adrian

closed 2016-05-15 by adrianbear