MaikStohn/UP3D #29: Deceleration / Retraction issue (or user calibration required)

adrianbear, 2016-05-10:
Hi @MaikStohn , @kscheff ,
I'm not sure if this is an issue or not. It is possible I just need to learn how to calibrate retraction settings on my printer.

But, with a standard preview from Simplify3d I can see a retraction point at this junction.
[image]
Of course, gcode does not show accelerations or decelerations, so I wont both adding a screenshot with the retraction rmoved.
But after converting to up code and then back to gcode (and removing the retraction point) I can see the deceleration (and acceleration).
[image]
In my real print, this is resulting in a disconnected tube (i.e. a C tube instead of an O tube).
[image]
The gap shown here is tough to get in a photo, but it runs the entire length of the print.

Is this an issue? Or is it just some retraction calibration settings I should amend?
If it's just retraction settings can you advise on a good technique to calibrate these?

Thanks
Adrian

MaikStohn, 2016-05-10:
Hi, 

retraction is only used to prevent "stringing" thin plastic lines coming out of nozzle during travel moves.

So missing material can only be if a travel move ended at the start point and re-retraction was to fast and no pressure was in nozzle. From a previous file I remember that you had a very very high retraction speed. Usually I use 20mm/sec and 1mm distance. UP software uses 5mm distance and also a very slow speed.

Maik

adrianbear, 2016-05-15:
I'm still not sure this is not an issue because the "gap" is not present if I print with version 0.3.0.
However I have found I can work around the problem by adding a retract distance of 2mm, an extra restart distance of 0.1mm and a retract speed of 1800mm/min. So I'm going to close the issue.
Thanks
Adrian

closed 2016-05-15 by adrianbear

MaikStohn, 2016-05-15:
Hi,

i made some test prints and also noticed a strange behavior.

I got a new idea and will rework the calculation.

Maik

MaikStohn, 2016-05-16:
There was a bug causing this in v0.6.x

I reimplemented the complete calculation based on discussion with @adrian and @kscheff in other issue.

v0.7.0 is now very very precise and also don't have this retract bug.

Maik