MaikStohn/UP3D #34: UP3DTOOLS on mac

peters5555, 2016-06-15:
Hi,
Not sure if it's just me but I get this error when trying to run the transcoder.

[image]

dfyx, 2016-06-15:
You have to run `./up3dtranscode` instead of just `up3dtranscode`. Otherwise OS X will look for the command in its standard paths instead of your local path. That's a general behavior of unix-like systems and has nothing to do with this project.

peters5555, 2016-06-15:
Thanks for your help. As you can tell I’m not a programmer, but love using 3D printers. Please let me know if there is another area I should be in to get help or give feedback using UP3DTOOLS.

Message protected by MailGuard: e-mail anti-virus, anti-spam and content filtering.
http://www.mailguard.com.au

Report this message as spam<https://console.mailguard.com.au/ras/1OE10XAvba/TXkRoK31gozjl9f0cp79L/8.174>

This e-mail and any attachments are intended for the addressee(s) only and may be confidential. Unauthorised use of this e-mail or its contents is strictly prohibited. If you are not the intended recipient please contact the sender as soon as possible by return e-mail and then please delete both messages. This notice should not be removed.
This e-mail and any attachments are intended for the addressee(s) only and may be confidential. Unauthorised use of this e-mail or its contents is strictly prohibited. If you are not the intended recipient please contact the sender as soon as possible by return e-mail and then please delete both messages. This notice should not be removed.

dfyx, 2016-06-15:
You're welcome and I'm sorry if my comment sounded harsh (English is not my native language).

I just wanted to make it clear that this is not a problem with UP3DTOOLS.

peters5555, 2016-06-16:
No worries:)
This e-mail and any attachments are intended for the addressee(s) only and may be confidential. Unauthorised use of this e-mail or its contents is strictly prohibited. If you are not the intended recipient please contact the sender as soon as possible by return e-mail and then please delete both messages. This notice should not be removed.

closed 2016-10-04 by MaikStohn