MaikStohn/UP3D #36: Cetus3D support

kscheff, 2016-09-11:
Hi, I have an Cetus3D prototype and it reports back a different PID.
System info tells me:
Cetus S7:
  Produkt-ID:	0x277f
  Hersteller-ID:	0x4745

So I modified up3dcomm.c to incorporate the new PID to see what it reports. Here is the result:
UP 3D PRINTER INFO:
TypeID:2710 Serial:1 ROM:6.1400 Model:Please update your app U1:1.00 U3:1 U4:102 U7:100
Max-X:-180.000000 Max-Y:180.000000 Max-Z:180.000000 Steps/mm-X:160.000000 Steps/mm-Y:160.000000 Steps/mm-Z:160.000000 Steps--A: 10.000000 Print sets: 1
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 05"
  ND:0.30 LT:0.05 SW:0.25 ST:1.00 HW:0.25 HS:4.00 HL:3.00 XW:0.28 XS:6.00 XL:3.00 SV:20.00 
  HV:25.00 XV:25.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.12 
  O1:20.00 O2:6.00 O3:1.00 O4:1000.00 O5:1500.00 O6:1500.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 

Not sure what all the information should tell me, but I like to build support for this this printer into the transcoder.

kscheff, 2016-09-12:
The issue might be related to the Slic3r settings. When using Simplify3D the sizes and speeds seem to match. I have set the extrusion multiplier in Simplify3D to 1.25, so there might bas till some differences on the extruder steps/mm despite the extruder itself looks very much the same to the UP Mini one.

Gitty-Hub, 2016-09-26:
So you can use the transcoder for the Cetus? Do you send data to it by typing cetus in stead of box, mini or plus?

kscheff, 2016-09-26:
@Gitty-Hub Yes, but this version is not yet released here

MaikStohn, 2016-10-04:
I added basic support for cetus today based on your findings.

Let's work out pull requests to merge more of your work.

Maik

closed 2016-10-04 by MaikStohn