MaikStohn/UP3D #37: Calculation of a for extruder

kscheff, 2016-09-12:
I have been looking for the extruder calculations and this confuses me. In the up3dconf.c file is the definition like this:
```
settings_t settings_mini = { 
  .steps_per_mm = { 854.0, 854.0, 854.0 },
  .max_rate = { 200, 200, 50 },
  .acceleration = { 1500, 1500, 300 },
  .junction_deviation = 0.05,
  .x_axes =  1, .y_axes =  0,
  .x_dir  =  1, .y_dir  = -1, .z_dir  = -1,
  .x_hspeed_hi = 50.0, .y_hspeed_hi = 50.0, .z_hspeed_hi = 50.0, .x_hofs_hi =  4.0, .y_hofs_hi =  4.0, .z_hofs_hi =  6.0,
  .x_hspeed_lo = 10.0, .y_hspeed_lo = 10.0, .z_hspeed_lo =  3.0, .x_hofs_lo =  9.0, .y_hofs_lo =  2.0, .z_hofs_lo =  2.0,
  .heatbed_wait_factor = 20.0,
};
```
So for the UP mini we have 854.0 steps per mm for XYZ axis. There is no definition for the extrude axis. Looking inside the hoststepper.c I see the following:
```
  //calc time based on corrected steps
  double tx = fabs(s_x / (512*settings.steps_per_mm[0]) / v);
  double ty = fabs(s_y / (512*settings.steps_per_mm[1]) / v);
  double ta = fabs(s_a / (512*settings.steps_per_mm[2]) / v);
```
The parameter .steps_per_mm[2] refers to the Z axis steps per mm, but it is used to calculate ta (which is for the extruder I guess).

It might be that on the UP mini and others the Z and A axis have the same steps_per_mm amount, but this might not be the cases on "newer" models.

kscheff, 2016-09-13:
I found a comment  for the A axis
```
up3ddata.h
  float    f_steps_mm_x;          //854.0   X steps/mm         (scale)
  float    f_steps_mm_y;          //854.0   Y steps/mm         (scale)
  float    f_steps_mm_z;          //854.0   Z steps/mm         (scale)
  float    f_unknown_a;           //40.0    A? some factor?    (ascale)             BUT in reality is 854.0 steps/mm
```
I used my calipers to measure the extrusion length when priming the nozzle with the UP software.  I measured 2 extrusions (a bit difficult to get a good reference point for the calipers and the filament) 75 mm and 77 mm. The Average is then 76 mm. Then I looked up the reported position of A in up3dshell. The extrusion uses exactly 20000 steps. So I get about 263 steps/mm for the extruder. This is way different than for the UP mini and co.

kscheff, 2016-09-15:
I now better calibrated the extruder of the Cetus3D by using a calibration g-code file which exactly extrudes 100 mm when executed. So I came up with a better steps/mm setting of 236.

This is the calibration g-code file I generated for this. For calibration adjustments the up3dtranscoder needs to be rebuild with new settings.

```
; extruder calibration for Cetus3D
;
; this file heats op the nozzle to 220°C
; and extrudes 100 mm with a speed of 
M107
M104 S220 ; set temperature
G28 ; home all axes
G1 Z120 F5000 ; lift nozzle
G1 X90 Y90 F1000 ; go to center position
M300 P100
M109 S220 ; wait for temperature to be reached
G21 ; set units to millimeters
G90 ; use absolute coordinates
M82 ; use absolute distances for extrusion
G92 E0

M300 P200            ; Beep

G1 E100 F100  ; extrude 100 mm

M300 P200            ; Beep 3 times as end
G4 P200
M300 P1
G4 P200
M300 P1

M104 S0        ; turn off hotted
G1 Z120 F5000  ; move nozzle up
```

MaikStohn, 2016-10-04:
Thanks for the analysis. I adopted your values and added basic cetus support.

closed 2016-10-04 by MaikStohn