MaikStohn/UP3D #42: New updates

kscheff, 2016-09-23:
Please note that new updates are currently available here:

https://github.com/UP3D-gcode/UP3D/releases

adrianbear, 2016-09-23:
I'll be right over and try some win testing today. 😊

Adrian Bear

https://github.com/UP3D-gcode/UP3D/releases

MaikStohn, 2016-10-04:
Hi, sorry for having no time...

I merged the pull requests and update to the same version number as kscheff now

dfyx, 2016-10-04:
@MaikStohn Would you consider either moving development over to the new organization or adding @kscheff as a developer? That way it would be easier to release bugfixes when you have no spare time.

Gitty-Hub, 2016-10-04:
Nice to see you back! No worries about the time, people have lives next to this :-)
It's great people care about this and development is still going.

MaikStohn, 2016-10-05:
Hi I offered kscheff to be developer in this repo but he is not sure if he really wants it.

I will continue to develop here in the original repository. 

Many new things to come soon :-)

Gitty-Hub, 2016-12-04:
Are there any developments to be released soon? Any news what you are working on, and will kscheff participate? I don't know what version to use now, is there a reliable version now for the Up Box that doesn't have dimensional issues? Tried some versiions but round models did not print well that's why I ask. Thanks for the development so far!

kscheff, 2016-12-04:
@gitty-hub could you be more detailed on the dimensional issues you are experiencing? If it's a X or Y scaling issue then we would need exact measurements, best compared to the original UP software to compensate for other setting issues.

Next you may try to verify if the UP Box extruder works correctly by messing e.g. 10cm extrude on the Filament side. Maybe there is an non-aligned setting in this software package.

Myself I don't have access to a UP Box and on the UP Mini and Cetus I don't see this issue. On my repro I am focusing on the Cetus and there is a macOS GUI in the download section.