MaikStohn/UP3D #43: plus - freezes after 28k

arhi, 2016-09-25:
I fixed the source to connect to 277D pid and not 2777 pid as originally set in code (for up2plus) and when I try to up3dload it freezes after 28.13kb
```
H:\toolpaths>dir
 Volume in drive H is DATA1
 Volume Serial Number is 2CFC-484F
[toolpaths.zip](https://github.com/MaikStohn/UP3D/files/491434/toolpaths.zip)

 Directory of H:\toolpaths

09/25/2016  01:45 AM    <DIR>          .
09/25/2016  01:45 AM    <DIR>          ..
09/25/2016  01:42 AM         2,448,813 melzi_highrise2.gcode
09/25/2016  01:45 AM         7,257,060 melzi_highrise2.umc
               2 File(s)      9,705,873 bytes
               2 Dir(s)  323,968,217,088 bytes free

H:\toolpaths>"e:\bin\UP3DTOOLS\up3dload - 7D.exe" melzi_highrise2.umc
^Cload: 28.13 kB sent (0%) [17.59 kB/sec]
H:\toolpaths>"e:\bin\UP3DTOOLS\up3dload - 7D.exe" melzi_highrise2.umc
UPload: 28.13 kB sent (0%) [17.96 kB/sec]

```

problematic files attached

kscheff, 2016-09-25:
@arhi could you please post up3dinfo results?

arhi, 2016-09-25:
```
e:\bin\UP3DTOOLS>"up3dinfo - 7D.exe"
UP 3D PRINTER INFO:
TypeID:277d Serial:91387 ROM:6.0900 Model:UP Plus 2(M) U1:1.00 U3:1 U4:102 U7:100
Max-X:-140.000000 Max-Y:140.000000 Max-Z:140.000000 Steps/mm-X:644.000000 Steps/mm-Y:644.000000 Steps/mm-Z:644.000000 Steps--A: 40.000000 Print sets:
6
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 0"
  ND:0.30 LT:0.15 SW:0.47 ST:1.00 HW:0.50 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:30.00
  HV:40.00 XV:40.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:0.75
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#1  Name:"Set 1"
  ND:0.30 LT:0.20 SW:0.50 ST:1.00 HW:0.52 HS:4.00 HL:3.00 XW:0.55 XS:6.00 XL:3.00 SV:35.00
  HV:45.00 XV:50.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.00
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:3000.00 O6:3000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#2  Name:"Set 2"
  ND:0.30 LT:0.25 SW:0.53 ST:1.00 HW:0.55 HS:6.00 HL:3.00 XW:0.57 XS:8.00 XL:3.00 SV:50.00
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.33
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#3  Name:"Set 3"
  ND:0.30 LT:0.30 SW:0.55 ST:1.00 HW:0.57 HS:6.00 HL:3.00 XW:0.60 XS:8.00 XL:3.00 SV:50.00
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:1.65
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#4  Name:"Set 4"
  ND:0.30 LT:0.35 SW:0.60 ST:1.00 HW:0.63 HS:6.00 HL:3.00 XW:0.66 XS:8.00 XL:3.00 SV:50.00
  HV:60.00 XV:70.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.10
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#5  Name:"Set 5"
  ND:0.30 LT:0.40 SW:0.65 ST:1.00 HW:0.68 HS:6.00 HL:3.00 XW:0.72 XS:8.00 XL:3.00 SV:50.00
  HV:60.00 XV:60.00 JS:150.00 SZ:1.00 HZ:1.05 SZ:0.90 FZ:2.60
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00

e:\bin\UP3DTOOLS>

```

arhi, 2016-09-25:
[image]