MaikStohn/UP3D #44: Work on New UpBox+ ?

3DWombat, 2017-01-08:
Does this tool work on the new UpBox+ . I think the machine ID is 2782

How do I make it work for this printer?

When I run it it comes up with ""[ERROR] USB Open Device (0001:2777/277D/277F/0001) not found" or something like that.

I got that number (2782)from the device manager properties for the usb device hardware id.

Regards Dennis

3DWombat, 2017-02-18:
Hello, anyone out there.

I would love to try this tool on my new UpBox+ but it seems it doesn't detect the device.

up3dinfo,exe comes back with "[ERROR] USB Open Device <4745:0001/2777/277D/277F> not found"

[image]

I assume from this picture the ID for the UpBox+ is 2782

Could someone add this USB id code to the repository, as I can't get it to compile on my machine(noob).

Regards Dennis

3DWombat, 2017-02-25:
Ok finally got something to compile. Ran up3dinfo and got this result.

>up3dinfo
UP 3D PRINTER INFO:
TypeID:2710 Serial:1 ROM:6.1500 Model:Please update your app U1:1.00 U3:1 U4:101
 U7:100
Max-X:6.000000 Max-Y:-3.000000 Max-Z:210.000000 Steps/mm-X:644.000000 Steps/mm-Y
:644.000000 Steps/mm-Z:3220.000000 Steps--A: 40.000000 Print sets: 1
U2:213.00 U3:-262.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 01"
  ND:0.00 LT:0.00 SW:0.00 ST:0.00 HW:0.00 HS:0.00 HL:0.00 XW:0.00 XS:0.00 XL:0.0
0 SV:0.00
  HV:0.00 XV:0.00 JS:0.00 SZ:0.00 HZ:1.00 SZ:0.90 FZ:0.45
  O1:20.00 O2:6.00 O3:1.00 O4:1500.00 O5:2000.00 O6:2000.00 U1:0.00 U2:0.00 U3:0
.00 U4:0.00 U5:0.00 U6:0.00

Do i need to update some info in the up files  for the "Model:Please update your app" line above?

I could get the up3dload to compile, but I haven't tested yet. up3dshell won't compile because of line 7 #include <ncurses/curses.h> in the upshell.c . I don't know if I got the libncurses installed properly. pacman came up with:---

 pacman -S mingw-w64-x86_64-ncurses
resolving dependencies...
looking for conflicting packages...

Packages (5) mingw-w64-x86_64-expat-2.2.0-2  mingw-w64-x86_64-gettext-0.19.7-1
             mingw-w64-x86_64-libsystre-1.0.1-2
             mingw-w64-x86_64-libtre-git-r122.c2f5d13-4
             mingw-w64-x86_64-ncurses-6.0.20161105-1

Total Download Size:    4.86 MiB
Total Installed Size:  31.24 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 mingw-w64-x86_64-ex...   120.3 KiB   118M/s 00:00 [#####################] 100%
 mingw-w64-x86_64-ge...     3.0 MiB   738K/s 00:04 [#####################] 100%
 mingw-w64-x86_64-li...    69.2 KiB  0.00B/s 00:00 [#####################] 100%
 mingw-w64-x86_64-li...     9.3 KiB  0.00B/s 00:00 [#####################] 100%
 mingw-w64-x86_64-nc...  1709.1 KiB   660K/s 00:03 [#####################] 100%
(5/5) checking keys in keyring                     [#####################] 100%
(5/5) checking package integrity                   [#####################] 100%
(5/5) loading package files                        [#####################] 100%
(5/5) checking for file conflicts                  [#####################] 100%
error: failed to commit transaction (conflicting files)
mingw-w64-x86_64-ncurses: /mingw64/lib/libncurses.a exists in filesystem
Errors occurred, no packages were upgraded.

So I renamed the file to orig_libncurses.a and ran pacman again.
It installed

Then I compiled up3dshell, it compiles with no errors.

I ran up3dshell, after first initializing the upbox+, and got some readouts, but the program won't quit on cntrl-c, I have to close the console window instead.

Are there any other files I need to update before trying this out? I will try and use it at a safe height , nowhere near the print bed first...

Dennis

3DWombat, 2017-02-25:
Output from up3dshell
[image]

Dennis

DiegoGiovany, 2017-05-07:
Any news on this? can you print some gcode? 

If I can ask,  please upload the compiled code/patch?  thanks !

reaganks, 2017-06-01:
Has anybody tried using Up Mini 2 for this system? I am sorry if this has been asked before.

ongspxm, 2018-05-23:
i got the `[ERROR] USB Open Device (0001:2777/277D/277F/0001) not found` error too.

this is my output of my lsusb
```
[  122.525798] usb 1-4: USB disconnect, device number 2
[  125.526869] usb 1-4: new full-speed USB device number 6 using xhci_hcd
[  125.668137] usb 1-4: New USB device found, idVendor=4745, idProduct=2782
[  125.668149] usb 1-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  125.668156] usb 1-4: Product: UP BOX+
[  125.668162] usb 1-4: Manufacturer: Tiertime
[ 125.668168] usb 1-4: SerialNumber: 531620
```

i am new to this, can i get some guide on how to recompile so that it works?