MaikStohn/UP3D #46: UP BOX - Nozzle height problem - Nozzle far from bed

brenorbm, 2017-01-17:
Hello! I am having problems setting the nozzle of my upbox to the right height from the bed. When the uploader send the transcoded GCODE, the nozzle start very, very far from the bed. Like 10 cm away or more! In my upbox studio, the nozzle height configured that I always use for best printing is 210.3 (latest MACOSX version of up studio). I am passing this value to the transcoder (the last argument of the command: 210.3).
I am also using the latest transcoder/uploader available to download (for MACOS). 
I am using the latest version of Slic3r to generate the GCODE. In slic3r I set for the bed x: 250; y: 200 and z offset as 0. The firmware I am using the default (RepRap/Marlin..)

Anyone has any idea???
Thanks a lot!

Gitty-Hub, 2017-01-17:
On my Up Box I have to add 25.5 mm to the Up software table height. So the Up Box can actually print quite some more in the Z height :-)

brenorbm, 2017-01-17:
Do you add this extra value in the parameter passed to the up3dtranscode? I see you add by 2.5 cm, but my extruder nozzle is about 10 cm away from the bed! Huge difference! Is this normal? Can I compensate this in the nozzle height parameter without cause any issues?

kscheff, 2017-01-17:
The nozzle height is the distance from the home position of the nozzle to the table. The Transcoder first homes all axises and then moves the table "nozzle_height" mm up before the first layer starts. Typically the first layer then moves the table one layer (or whatever you have set in the Slicer) down before extrusion starts. 

If your print starts 10cm above the table, you should increase nozzle_height. Be careful not to drive the table into the nozzle. A good practice is to increase nozzle_height step-by-step, e.g. 10 mm. The closer the nozzle gets to the table the smaller the steps should be!

brenorbm, 2017-01-18:
Thanks a lot kscheff! Solved the issue here. I did that procedure you told me. I am getting perfect prints now after figuring out the correct settings for the first layer to my Up! Box. Just to share with you, 0.2mm height for the first layer is working perfect!

closed 2017-01-19 by brenorbm