MaikStohn/UP3D #47: My Mini starts to print without heating the nozzle!

putzvarruckt, 2017-03-23:
After gerenrating a gcode file with Slic3r (Temperature of Extruder set to 228° and Bed to 90°) I checked the file and found M104 S228 command lines in the gcode file.
Transcoding works fine, uploading works fine, after a few seconds the printer initializes and starts to move.
The temperature of the nozzle stays calm. 
Any ideas?

Thanky in advance...

Fardenco, 2017-04-20:
This might be an issue with your slicer settings
In Simplify3d for example there is a checkbox "Wait for temperature controller to stabilize before beginning build"
[image]

You should open the gcode and check the the heating commands

putzvarruckt, 2017-04-20:
Thank you!
After some trail and error the printer works now. 
Mainly I was to impatient. The printer first lets the bed getting warm for about 10 Minutes.
I thought that it does nothing...

But now as I know I have to wait and I know the M104 and M109 codes I got it running...

Thank you again

Friedrich

Fardenco, 2017-04-20:
For the bed you could preheat it via the standard UP! software while preparing your stl/gcode or anything, and then just use M104, for heating without waiting the 10 minutes for the print to start

putzvarruckt, 2017-04-20:
Yes would be possible.
Do you have a profile set for Simplify3D that fits the UP!Mini?

Fardenco, 2017-04-20:
Right now I'm playing with differents settings to try to figure out what could cause the crashes reported [here](https://github.com/MaikStohn/UP3D/issues/26)
I'm trying to get useful datas to provide to devs so they can figure out which part of the code is causing this.
I would really need this issue to be resolved since it's the only real wall to cross before unleaching my printer !

Anyway the settings are really basics, I've only set the printing area to 120*120*120, nozzle diameter and extrusion width to 0.4, and temperature settings.
I'll upload the profile when the issue will be resolved

putzvarruckt, 2017-04-21:
Why are you using 0,4 nozzle diameter? Imo the printe has 0,3!

Fardenco, 2017-04-21:
I'm pretty sure the default nozzle of UP! printers is .4mm
https://www.up3d.com/community/viewtopic.php?t=4971
But maybe you changed yours for a smaller one

putzvarruckt, 2017-04-21:
Strange, why is the setting in Fixup3D telling me it is a 0.3 nozzle?
That my also be the reason for my poor printing quality out of Slic3r vs. Up3D Software....
I will try 0.4 in Slic3r and Simplify3D...

siraeryn, 2018-01-14:
I have the same issue as the OP @putzvarruckt 
Notice the temperature under Heater 2 is a little above your set nozzle temperature.  This is not a matter of waiting for the nozzle to heat up.  If it was starting to heat, the temp would start to increase as soon as the print starts, but it doesn't.
For whatever reason, the code to the printer appears to be heating something else.  If you close the shell and open UPStudio, you can see that the nozzle temperature is steady.  Mine was sitting at about 34 degrees.
So does the UP Mini have more than one heater?  Heater 3 appears to be the bed.  What are 2 and 4?

arendtio, 2020-08-19:
Yesterday, I was trying to find a way around the 'printer stops after a few minutes/layers' bug and tried different G-Code Type settings within 'PrusaSlicer' for that matter. When I used the setting Mach3/LinuxCNC I also had the issue that the printer started to print without heating the nozzle. Maybe it is supposed to work that way, but I found it curious, that it was exactly the behavior stated in this bug report. When using RepRap/Sprinter instead, this bug didn't appear.