MaikStohn/UP3D #50: Cetus ROM 6.14 support

yetifrisstlama, 2017-12-16:
looks like there were some firmware changes in the latest cetus 3D model. 

```bash
./up3dload ../test.umc 
UPload: 1.07 kB sent (100%) [440.43 kB/sec]
ERROR: Write to SD not finished
```

I tried to comment out that check with the result of the printer slamming its bed against the upper limit.

Also I get some interesting output from up3dinfo

```bash
 ./up3dinfo 
UP 3D PRINTER INFO:
TypeID:2710 Serial:1 ROM:6.1400 Model:Please update your app U1:1.00 U3:1 U4:102 U7:100
Max-X:-180.000000 Max-Y:180.000000 Max-Z:185.000000 Steps/mm-X:160.000000 Steps/mm-Y:160.000000 Steps/mm-Z:160.000000 Steps--A: 10.000000 Print sets: 1
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 05"
  ND:0.00 LT:0.00 SW:0.00 ST:0.00 HW:0.00 HS:0.00 HL:0.00 XW:0.00 XS:0.00 XL:0.00 SV:0.00 
  HV:0.00 XV:0.00 JS:0.00 SZ:0.00 HZ:1.00 SZ:0.90 FZ:0.12 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
```

I'm willing to work on this but haven't quite figured out yet where to start.

Cheers,
Michael

etet100, 2018-05-01:
Hi. I found a way to fix it:

In up3d.c change:

`static const uint8_t UP3D_CMD_52[] = { 0x52, 0x00 }; //GetPrinterInfo`

to 

`static const uint8_t UP3D_CMD_52[] = { 0x52, 0x77 }; //GetPrinterInfo`

And

```
  if( pidata->u32_NumSets>8 )
    return false;
```

to

```
  if( pidata->u32_NumSets>9 )
    return false;
```

And voila!

```
UP 3D PRINTER INFO:
TypeID:277f Serial:70172052 ROM:6.1400 Model:Cetus S7(NL) U1:1.00 U3:1 U4:102 U7:100
Max-X:-180.000000 Max-Y:180.000000 Max-Z:185.000000 Steps/mm-X:160.000000 Steps/mm-Y:160.000000 Steps/mm-Z:160.000000 Steps--A: 10.000000 Print sets: 9
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 05"
  ND:0.30 LT:0.05 SW:0.25 ST:1.00 HW:0.25 HS:4.00 HL:3.00 XW:0.30 XS:6.00 XL:3.00 SV:20.00
  HV:25.00 XV:25.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.12
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00
Set#1  Name:"Set 07"
  ND:0.30 LT:0.07 SW:0.27 ST:1.00 HW:0.30 HS:4.00 HL:3.00 XW:0.33 XS:6.00 XL:3.00 SV:25.00
  HV:25.00 XV:30.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.19

```

Samuel-Clarke123, 2018-05-16:
Hey etet100, I've compiled from source, following your specifications, but whilst running up3dinfo, im getting some strange errors.

Here's my output:
```
UP 3D PRINTER INFO:
TypeID:277f Serial:70174067 ROM:6.1400 Model:Cetus S7(NL) U1:1.00 U3:1 U4:102 U7:100
Max-X:-180.000000 Max-Y:180.000000 Max-Z:185.000000 Steps/mm-X:160.000000 Steps/mm-Y:160.000000 Steps/mm-Z:160.000000 Steps--A: 10.000000 Print sets: 9
U2:0.00 U3:0.00 U10:0.00 U11:0.00 U12:0.00 U13:1.00
Set#0  Name:"Set 05"
  ND:0.30 LT:0.05 SW:0.25 ST:1.00 HW:0.25 HS:4.00 HL:3.00 XW:0.30 XS:6.00 XL:3.00 SV:20.00 
  HV:25.00 XV:25.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.12 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#1  Name:"Set 07"
  ND:0.30 LT:0.07 SW:0.27 ST:1.00 HW:0.30 HS:4.00 HL:3.00 XW:0.33 XS:6.00 XL:3.00 SV:25.00 
  HV:25.00 XV:30.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.19 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#2  Name:"Set 10"
  ND:0.30 LT:0.10 SW:0.30 ST:1.00 HW:0.32 HS:4.00 HL:3.00 XW:0.35 XS:6.00 XL:3.00 SV:25.00 
  HV:25.00 XV:30.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.30 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#3  Name:"Set 15"
  ND:0.30 LT:0.15 SW:0.40 ST:1.00 HW:0.40 HS:4.00 HL:3.00 XW:0.45 XS:6.00 XL:3.00 SV:30.00 
  HV:40.00 XV:40.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:1.01 FZ:0.60 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#4  Name:"Set 1"
  ND:0.30 LT:0.20 SW:0.45 ST:1.00 HW:0.45 HS:4.00 HL:3.00 XW:0.50 XS:6.00 XL:3.00 SV:35.00 
  HV:45.00 XV:50.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:0.90 FZ:0.90 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#5  Name:"Set 2"
  ND:0.30 LT:0.25 SW:0.50 ST:1.00 HW:0.50 HS:6.00 HL:3.00 XW:0.55 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:1.01 FZ:1.25 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#6  Name:"Set 3"
  ND:0.30 LT:0.30 SW:0.55 ST:1.00 HW:0.55 HS:6.00 HL:3.00 XW:0.60 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:1.01 FZ:1.65 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#7  Name:"Set 4"
  ND:0.30 LT:0.35 SW:0.60 ST:1.00 HW:0.60 HS:6.00 HL:3.00 XW:0.65 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:70.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:1.01 FZ:2.10 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 
Set#8  Name:"Set 5"
  ND:0.30 LT:0.40 SW:0.65 ST:1.00 HW:0.65 HS:6.00 HL:3.00 XW:0.70 XS:8.00 XL:3.00 SV:50.00 
  HV:60.00 XV:60.00 JS:180.00 SZ:1.00 HZ:1.00 SZ:1.01 FZ:2.60 
  O1:20.00 O2:6.00 O3:1.00 O4:3000.00 O5:6000.00 O6:6000.00 U1:0.00 U2:0.00 U3:0.00 U4:0.00 U5:0.00 U6:0.00 

*** stack smashing detected ***: ./up3dinfo terminated
Aborted
```

Could you help me at all? Thanks.

etet100, 2018-05-16:
Hi. I overlooked this:

` TT_tagPrinterInfoSet    pisets[8];`

Not enought space for 9 sets. Change it to 9 or 10.

Samuel-Clarke123, 2018-05-17:
Hey, I tried, and now upinfo works! great!
One problem with the upload however.

```
ned_flanders@paint ~/Desktop/3dprinter/UP3D/UP3DTOOLS $ sudo ./up3dload ./test_skull.umc 
UPload: 772.81 kB sent (100%) [66.42 kB/sec]
ERROR: Write to SD not finished
ned_flanders@paint ~/Desktop/3dprinter/UP3D/UP3DTOOLS $ ./make.sh
ned_flanders@paint ~/Desktop/3dprinter/UP3D/UP3DTOOLS $ sudo ./up3dload ./test_skull.umc 
UPload: 772.81 kB sent (100%) [69.85 kB/sec]
```

Then, after commenting it out, the light goes red, and the print bed slams backwards. Any help?

etet100, 2018-05-17:
Personally i think it's a waste of time. Project seems to be abandoned quite long time ago. I't very hard to analyze all these packets and protocols. 

I'm working on alternative cpu board. It's very cheap (4$ maple mini board) and it's running open source marlin firmware.

arhi, 2018-05-17:
@etet100 why don't you look at already existing cpumod board with nxp that runs smoothieware, marlin is crap, this is way better solution, the board already exist, you can improve on it if you wish.. the pinout and everything is exposed, you can see my images on up3d forum with pinout of the extruder board and config of all the pins on the "cpu board"..

etet100, 2018-05-18:
You mean tinyfab board? I was going to buy it but it's too expensive. What is wrong with marlin? It is working fine. I have just finished 12 hours printing, i have sensorless homing (which is not supported by tinyfab), i'm working on bltouch support. 

[image]

Samuel-Clarke123, 2018-05-18:
It's a shame to hear how dead all of this is. I was working with the cetus3d on linux and just really desperately wanted a way to print with it, instead of having to buy some stupid windows laptop to run up studio. I got one with me, so I guess it's alright, just wish there was some decent support.

arhi, 2018-05-18:
well the latest version of "studio" software support import of the g-code so you in theory can use original electronics with any of the g-code generators (craftware, s3d, slic3r, cura...) these days so it's not that important to change the electronics.. 

tinyfab cpu board with few minor changes would be superb solution imo

etet100, 2018-05-19:
I know i can import g-code. There is a reason why i used it only once. 

And i still think that cheap stm32 board is a nice project. I don't know what is so great about tinyfab boards.

arhi, 2018-05-19:
never tried it myself (gcode import) as it was added way after i replaced the board

tinyfab is nothing special, it's smoothieware that's awesome

Samuel-Clarke123, 2018-05-20:
Hooked up an old laptop with a pentium in it that I was never going to use, cored out a cupboard, put all the components into it. I now teamviewer into that laptop. There's a viewing port from one of the cupboard drawers, and the other to remove the machine, like a cupboard server rack. It has an arduino controlling automatic lighting for whenever I open it, and is fairly quiet now.
Thanks for the help everybody, even if I didnt end up using UP3D tools.